google-deepmind/mujoco_mpc #23: AddressSanitizer issue: residual parameters array and the UI

nimrod-gileadi, 2022-12-05:
When running the main binary under ASAN, there's a reliable crash:

1. Launch binary.
2. Switch from Humanoid to Quadruped Flat.
3. As soon as the quadruped shows up, but before the simulation is automatically started, tap space.

yuvaltassa, 2023-01-27:
Is this still an issue?

nimrod-gileadi, 2023-02-06:
Yep.

scakki, 2023-06-12:
Hello,
I am getting a segmentation error every time I switch from any task to Quadruped Flat Task. All other tasks seem to be working fine, except for Quadruped Flat. How can I fix this?
Thank you

yuvaltassa, 2023-06-12:
If you run with `--task="Quadruped Flat"` does this still happen, or only when switching in the UI?

scakki, 2023-06-12:
Yes. It happens both ways

Lhumd, 2023-10-11:
I had the same issue. I updated my python to 3.11, and now it works!

scakki, 2023-10-11:
Thank you for your response. This helps.

thowell, 2024-01-12:
@nimrod-gileadi is this still an issue? 

I just tried it and did not experience the failure case. My steps to replicate:
`cmake -G Ninja -DCMAKE_BUILD_TYPE=ASAN`
`cmake --build . --config=Release --target mjpc`
`./mjpc`
Follow 1-3 above