google-deepmind/mujoco_mpc #35: build error: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)

IcarusWizard, 2022-12-15:
Hi, thanks for the great work. I want to try out the software and see whether it can help me to generate some datasets for my research. However, I encounter the error `no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)` when building with `gcc-10`. Could you please help me solve the problem? Thanks!

The complete log is following:
```
[main] Building folder: mujoco_mpc 
[main] Configuring folder: mujoco_mpc 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/gcc-10 -S/home/xingyuan/mujoco_mpc/mjpc -B/home/xingyuan/mujoco_mpc/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is GNU 10.3.0
[cmake] -- The CXX compiler identification is GNU 9.4.0
[cmake] -- Check for working C compiler: /usr/bin/gcc-10
[cmake] -- Check for working C compiler: /usr/bin/gcc-10 -- works
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Check for working CXX compiler: /usr/bin/c++
[cmake] -- Check for working CXX compiler: /usr/bin/c++ -- works
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Performing Test CAN_BUILD_AVX
[cmake] -- Performing Test CAN_BUILD_AVX - Success
[cmake] -- Performing Test SUPPORTS_LLD
[cmake] -- Performing Test SUPPORTS_LLD - Failed
[cmake] -- Performing Test SUPPORTS_GOLD
[cmake] -- Performing Test SUPPORTS_GOLD - Success
[cmake] -- Performing Test SUPPORTS_GC_SECTIONS
[cmake] -- Performing Test SUPPORTS_GC_SECTIONS - Success
[cmake] -- Looking for pthread.h
[cmake] -- Looking for pthread.h - found
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
[cmake] -- Looking for pthread_create in pthreads
[cmake] -- Looking for pthread_create in pthreads - not found
[cmake] -- Looking for pthread_create in pthread
[cmake] -- Looking for pthread_create in pthread - found
[cmake] -- Found Threads: TRUE  
[cmake] CHECK_STARTmujoco::FindOrFetch: checking for targets in package `mujoco`
[cmake] CHECK_FAILtarget `mujoco::mujoco` not defined.
[cmake] CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `mujoco`
[cmake] CHECK_STARTmujoco::FindOrFetch: checking for targets in package `qhull`
[cmake] CHECK_FAILtarget `qhull` not defined.
[cmake] CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `qhull`
[cmake] -- Looking for sys/types.h
[cmake] -- Looking for sys/types.h - found
[cmake] -- Looking for inttypes.h
[cmake] -- Looking for inttypes.h - found
[cmake] -- Looking for stddef.h
[cmake] -- Looking for stddef.h - found
[cmake] -- Looking for stdint.h
[cmake] -- Looking for stdint.h - found
[cmake] -- Check size of off_t
[cmake] -- Check size of off_t - done
[cmake] -- Looking for fseeko
[cmake] -- Looking for fseeko - found
[cmake] -- Looking for ftello
[cmake] -- Looking for ftello - found
[cmake] -- Looking for PRIdMAX
[cmake] -- Looking for PRIdMAX - found
[cmake] -- 
[cmake] -- ========== qhull Build Information ==========
[cmake] -- Build Version:                             8.1-alpha1
[cmake] -- Install Prefix (CMAKE_INSTALL_PREFIX):     /usr/local
[cmake] -- Binary Directory (BIN_INSTALL_DIR):        bin
[cmake] -- Library Directory (LIB_INSTALL_DIR):       lib
[cmake] -- Include Directory (INCLUDE_INSTALL_DIR):   include
[cmake] -- Documentation Directory (DOC_INSTALL_DIR): share/doc/qhull
[cmake] -- Man Pages Directory (MAN_INSTALL_DIR):     share/man/man1
[... 275 lines cut ...]
[build]       |        ^~~~~~~~
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const mjuiDef_&’
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(mjuiDef_&&)’
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘mjuiDef_&&’
[build] /home/xingyuan/mujoco_mpc/mjpc/agent.cc:312:14: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)
[build]   312 |         "0 1"};
[build]       |              ^
[build] In file included from /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mujoco.h:40,
[build]                  from /home/xingyuan/mujoco_mpc/mjpc/agent.h:23,
[build]                  from /home/xingyuan/mujoco_mpc/mjpc/agent.cc:15:
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(const mjuiDef_&)’
[build]   303 | struct mjuiDef_ {                 // table passed to mjui_add()
[build]       |        ^~~~~~~~
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const mjuiDef_&’
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(mjuiDef_&&)’
[build] /home/xingyuan/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:303:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘mjuiDef_&&’
[build] [ 79%] Building CXX object CMakeFiles/libmjpc.dir/tasks/swimmer/swimmer.cc.o
[build] [ 80%] Building CXX object CMakeFiles/libmjpc.dir/tasks/walker/walker.cc.o
[build] make[2]: *** [CMakeFiles/libmjpc.dir/build.make:89: CMakeFiles/libmjpc.dir/agent.cc.o] Error 1
[build] make[2]: *** Waiting for unfinished jobs....
[build] make[1]: *** [CMakeFiles/Makefile2:1249: CMakeFiles/libmjpc.dir/all] Error 2
[build] make: *** [Makefile:141: all] Error 2
[proc] The command: /usr/bin/cmake --build /home/xingyuan/mujoco_mpc/build --config Debug --target all -j 10 -- exited with code: 2 and signal: null
[build] Build finished with exit code 2
```

kevinzakka, 2022-12-15:
Are you building with C++ 20?

saran-t, 2022-12-15:
Also you're using different compilers for C++ and C.

```[cmake] -- The C compiler identification is GNU 10.3.0
[cmake] -- The CXX compiler identification is GNU 9.4.0```

IcarusWizard, 2022-12-16:
Thanks for the help, @kevinzakka @saran-t. Following the GitHub action, I switch to `clang-12` and the code can be compiled correctly!

FYI, I fix the mismatch of the version between the C compiler and the C++ compiler, and it still raises the same error. I also try GNU 8, and GNU 9, and the error stays the same.  Could be some bugs in the compiler.

closed 2022-12-16 by IcarusWizard