google-deepmind/mujoco_mpc #43: Failed to build target "mjpc" in release mode

dtch1997, 2023-01-13:
Following the documentation, I configured VSCode to build target 'mjpc' in Release mode, but the build failed with the following error: 

```
[build] /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc: In member function ‘void mjpc::Agent::Gui(mjUI&)’:
[build] /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:291:57: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)
[build]   291 |                           DataAt(task_.weight, i), "0 1"};
[build]       |                                                         ^
[build] In file included from /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mujoco.h:40,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.h:23,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:15:
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(const mjuiDef_&)’
[build]   309 | struct mjuiDef_ {                 // table passed to mjui_add()
[build]       |        ^~~~~~~~
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const mjuiDef_&’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(mjuiDef_&&)’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘mjuiDef_&&’
[build] /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:310:74: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)
[build]   310 |     defFeatureParameters[0] = {mjITEM_SEPARATOR, "Residual Parameters", 1};
[build]       |                                                                          ^
[build] In file included from /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mujoco.h:40,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.h:23,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:15:
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(const mjuiDef_&)’
[build]   309 | struct mjuiDef_ {                 // table passed to mjui_add()
[build]       |        ^~~~~~~~
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const mjuiDef_&’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(mjuiDef_&&)’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘mjuiDef_&&’
[build] /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:315:14: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)
[build]   315 |         "0 1"};
[build]       |              ^
[build] In file included from /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mujoco.h:40,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.h:23,
[build]                  from /home/daniel/Documents/github/mujoco_mpc/mjpc/agent.cc:15:
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(const mjuiDef_&)’
[build]   309 | struct mjuiDef_ {                 // table passed to mjui_add()
[build]       |        ^~~~~~~~
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const mjuiDef_&’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note: candidate: ‘constexpr mjuiDef_& mjuiDef_::operator=(mjuiDef_&&)’
[build] /home/daniel/Documents/github/mujoco_mpc/build/_deps/mujoco-src/include/mujoco/mjui.h:309:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘mjuiDef_&&’
[build] [ 91%] Building CXX object CMakeFiles/libmjpc.dir/planners/sampling/policy.cc.o
[build] [ 91%] Building CXX object CMakeFiles/libmjpc.dir/planners/gradient/gradient.cc.o
[build] [ 91%] Building CXX object CMakeFiles/libmjpc.dir/planners/gradient/planner.cc.o
[build] [ 91%] Linking CXX static library ../../../../lib/libabsl_flags_internal.a
[build] make[3]: *** [CMakeFiles/libmjpc.dir/build.make:89: CMakeFiles/libmjpc.dir/agent.cc.o] Error 1
[build] make[3]: *** Waiting for unfinished jobs....
[build] [ 91%] Built target absl_flags_internal
[build] make[2]: *** [CMakeFiles/Makefile2:1249: CMakeFiles/libmjpc.dir/all] Error 2
[build] make[2]: *** Waiting for unfinished jobs....
[build] [ 91%] Linking CXX static library ../../../../lib/libabsl_cord.a
[build] [ 91%] Built target absl_cord
[build] make[1]: *** [CMakeFiles/Makefile2:1342: CMakeFiles/mjpc.dir/rule] Error 2
[build] make: *** [Makefile:189: mjpc] Error 2
[proc] The command: /usr/bin/cmake --build /home/daniel/Documents/github/mujoco_mpc/build --config Release --target mjpc -j 18 -- exited with code: 2 and signal: null
[build] Build finished with exit code 2
```

thowell, 2023-01-13:
Please see #35. Note, we are using `clang-14` with Github Actions.

dtch1997, 2023-01-16:
Thank you! I eventually managed to build the project. 

On Ubuntu 22.04, first ensure gcc 11 is installed: 
```
$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
```

Then build the project from command-line: 
```
cd mjpc 
mkdir build
cmake .. -DCMAKE_BUILD_TYPE Release
make -j 16
```
Finally run `./bin/mjpc` which opens the program.

closed 2023-01-16 by dtch1997