google-deepmind/mujoco_mpc #46: Add documentation about how to use GUI

dtch1997, 2023-01-16:
Thanks very much for releasing this amazing software! At the moment, there is no explanation on how to use the program, but I was able to deduce the following via (a) listening to the talk, and (b) trial-and-error: 

1. When you open the program, there are no actions applied by default. To actually enable actions, we must click on Agent --> (Planner, Action) to enable planning and execution of actions. 
2. The camera can be changed with mouse control. Left-click + drag changes view orientation, right-click + drag performs translation, scroll to zoom in/out. 
3. In 'Option', clicking on the various buttons brings up various useful on-screen displays. For example, 'Right UI' turns on a display of all control DoFs. Worth clicking and experimenting. 
4. In 'Task', we can configure the optimization process. Meaning of risk, cost weights, residual parameters are explained in the talk and paper.  

The remaining categories are pretty self-explanatory IMO. 

Unresolved questions:
A: In the talk, it is shown that Yuval can drag / drop objects around by clicking on the UI, but I wasn't able to discover how this is done
B: What does 'Group enable' do?

yuvaltassa, 2023-01-16:
Hi Daniel, thanks for this.

* Many of the GUI actions you mention are simply part of MuJoCo's native [simulate](https://mujoco.readthedocs.io/en/latest/programming/samples.html#sasimulate) viewer. In simulate there is a help pane that comes up with the F1 key. Does this not happen in MJPC?
* MJPC adds three keyboard shortcuts:
  1. The `Enter` key starts and stops the planner.
  2. The `\` key starts and stops the controller (sending actions from the planner to the model).
  3. The `9` key turns the traces on/off.
* I thought we did have a GUI user guide? Maybe it's in a draft state that was never submitted. @kevinzakka, @thowell, do you remember such a guide.
* Regarding your specific questions. These are regular `simulate` usage shortcuts. Double-click on a body to select it, then `Ctrl`-{left/right} drag, either in run mode or pause mode. I recommend playing around with `simulate` a little bit to get the hang of it.

closed 2023-01-16 by dtch1997

yuvaltassa, 2023-01-16:
Hi Daniel,

I think that the [shift] is in brackets to denote that it's optional. Without the shift the motion plane vertical, with the shift it's horizontal.

If you care to improve the documentation, please feel free to send us a pull request!