google-deepmind/mujoco_mpc #52: Could mjpc be used to collect dataset?

IcarusWizard, 2023-01-18:
Hey, I am working on an imitation learning project and I find mjpc can generate some decent trajectories for some tasks of interest. So I wonder if I can export trajectories from mjpc to build a dataset?

thowell, 2023-01-18:
Currently, MJPC does not support saving trajectories from the GUI. 

For now, you could write a script that utilizes planners in a [headless fashion](https://github.com/deepmind/mujoco_mpc/blob/9897673993c09e892a75409813ec9693b4b3c3ee/mjpc/test/agent/agent_test.cc#L92) to save trajectories.

IcarusWizard, 2023-01-19:
Thanks for the quick reply!

I see. Writing C++ directly would also work, with a bit more effort. Haven't written C++ for years, but it gonna be fun.

closed 2023-01-19 by IcarusWizard