google-deepmind/mujoco_mpc #109: Dependency files needed change to void compiling error

gaolongsen, 2023-05-03:
Dear Deepmind Team, after I cmake the project, I found some issues when I tried to use this lib on my Ubuntu20 system. I ask someone to do the same test and the problem are the same. So I would like to let you know and make some changes for that:
1. **mujoco_mpc/cmake/MpcOptions.cmake** and **/home/wam/mujoco_mpc/_deps/mujoco-build/CMakeFiles/mujoco.dir/flags.make** I think you should remove all **-Werror** to make sure that even though some warring occurred, users can also compile the project.
2. **mujoco_mpc/mjpc/agent.cc** add **`(struct mjuiDef_)`** in `line 366`, `line 386` and `line 389` to make sure that no errors occur during the assignment of structure variables.

saran-t, 2023-05-04:
If you get an error from the `-Werror` then it usually indicates that there's a real issue with the code that needs to be fixed. Can you tell us what the error is?

gaolongsen, 2023-05-04:
But so weird after I `cmake .` and `cmake --build .` again, no error as what I listed above.

saran-t, 2023-05-04:
You're somehow building tests from gRPC... those aren't coming from this repo's code.

erwincoumans, 2023-05-10:
Same issue here, on Mac M1. The target mjpc compiles and runs fine, so it seems harmless.

```
erwincoumans@Erwins-MBP mujoco_mpc % make
[  0%] Built target absl_spinlock_wait
[  0%] Built target absl_log_severity
[  0%] Built target absl_raw_logging_internal
[  0%] Built target absl_base
[  0%] Built target threadpool
[  0%] Built target ccd
[  0%] Built target qhullstatic_r
[  0%] Built target lodepng
[  0%] Built target tinyxml2
[  0%] Built target tinyobjloader
[  2%] Built target mujoco
[  2%] Built target absl_malloc_internal
[  2%] Built target absl_throw_delegate
[  2%] Built target absl_time_zone
[  2%] Built target absl_debugging_internal
[  2%] Built target absl_stacktrace
[  3%] Built target absl_strings_internal
[  3%] Built target absl_demangle_internal
[  3%] Built target absl_int128
[  3%] Built target absl_strings
[  3%] Built target absl_symbolize
[  3%] Built target absl_exponential_biased
[  3%] Built target absl_graphcycles_internal
[  3%] Built target absl_civil_time
[  3%] Built target absl_time
[  3%] Built target absl_synchronization
[  3%] Built target absl_hashtablez_sampler
[  3%] Built target absl_bad_optional_access
[  3%] Built target absl_raw_hash_set
[  3%] Built target absl_crc_cpu_detect
[  3%] Built target absl_crc_internal
[  3%] Built target absl_crc32c
[  3%] Built target absl_crc_cord_state
[  3%] Built target absl_bad_variant_access
[  3%] Built target absl_city
[  3%] Built target absl_low_level_hash
[  3%] Built target absl_hash
[  3%] Built target absl_random_distributions
[  3%] Built target absl_random_seed_gen_exception
[  3%] Built target absl_random_internal_seed_material
[  3%] Built target absl_random_internal_platform
[  3%] Built target absl_random_internal_randen_hwaes_impl
[  3%] Built target absl_random_internal_randen_slow
[  3%] Built target absl_random_internal_randen_hwaes
[  4%] Built target absl_random_internal_randen
[  4%] Built target absl_random_internal_pool_urbg
[  4%] Built target absl_random_seed_sequences
[  4%] Built target absl_cord_internal
[  5%] Built target absl_cordz_functions
[  5%] Built target absl_cordz_handle
[  5%] Built target absl_cordz_info
[  5%] Built target absl_cord
[  7%] Built target libmjpc
[  7%] Copying tasks into binary directory
[  7%] Built target copy_resources
[  7%] Copying menagerie assets into binary directory
[  7%] Built target copy_menagerie_resources
[  7%] Built target platform_ui_adapter
[  8%] Built target glfw
[  8%] Built target absl_flags_program_name
[  8%] Built target absl_flags_config
[  9%] Built target absl_str_format_internal
[  9%] Built target absl_flags_marshalling
[  9%] Built target absl_flags_commandlineflag_internal
[  9%] Built target absl_flags_commandlineflag
[  9%] Built target absl_flags_private_handle_accessor
[... 159 lines cut ...]
      _test_large_inflate in example.o
      _test_sync in example.o
      _test_dict_inflate in example.o
  "_inflateInit_", referenced from:
      _test_inflate in example.o
      _test_large_inflate in example.o
      _test_sync in example.o
      _test_dict_inflate in example.o
  "_inflateSetDictionary", referenced from:
      _test_dict_inflate in example.o
  "_inflateSync", referenced from:
      _test_sync in example.o
  "_uncompress", referenced from:
      _test_compress in example.o
  "_zlibCompileFlags", referenced from:
      _main in example.o
  "_zlibVersion", referenced from:
      _main in example.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/example] Error 1
make[1]: *** [_deps/grpc-build/third_party/zlib/CMakeFiles/example.dir/all] Error 2
make: *** [all] Error 2

```

closed 2024-01-12 by thowell