google-deepmind/mujoco_mpc #112: ERROR: mjui_add: invalid shortcut specification + segmentation fault

poppy-alpaca, 2023-05-23:
Hello, 

I just tried building the binary and ran into an issue, using both VSCode and manually on the terminal. 

I am using an M1 Macbook (macOS 13.4) and
```
clang++ --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.5.0
Thread model: posix
```
When I build and run the release binary I get
```
>> mujoco_mpc/build/bin % ./mjpc 
MuJoCo version 2.3.6
Hardware threads: 8
Agent threads: 5
ERROR: mjui_add: invalid shortcut specification

Press Enter to exit ...
zsh: segmentation fault  ./mjpc
```
I built mjpc using the Github workflow commands. I am also able to build and run the current MuJoCo GUI. 

Below I will also include the crash report.
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               mjpc [13808]
Path:                  /Users/USER/*/mjpc
Identifier:            mjpc
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [13791]
Responsible:           Terminal [1257]
User ID:               501

Date/Time:             2023-05-23 22:39:37.9105 +0200
OS Version:            macOS 13.4 (22F66)
Report Version:        12
Anonymous UUID:        5910434B-24D2-83F4-94C0-58C26FE7D4BE

Time Awake Since Boot: 2700 seconds

System Integrity Protection: enabled

Crashed Thread:        6

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000006ba4e0
Exception Codes:       0x0000000000000001, 0x00000000006ba4e0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [13808]

VM Region Info: 0x6ba4e0 is not in any region.  Bytes before following region: 4290632480
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100298000-100340000    [  672K] r-x/r-x SM=COW  ...s/USER/*/mjpc

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1964aeb9c madvise + 8
1   libsystem_malloc.dylib        	       0x196318654 free_large + 776
2   libsystem_malloc.dylib        	       0x19632267c _szone_free + 720
3   libmujoco.2.3.6.dylib         	       0x100617024 mj_deleteModel + 28
4   mjpc                          	       0x1002d6c54 mjpc::Agent::~Agent() + 36
5   mjpc                          	       0x1002d708c mujoco::Simulate::~Simulate() + 204
[... 440 lines cut ...]
      "deploymentId" : 240000276
    },
    {
      "rolloutId" : "62699e1ec1ff2978b47f6c3b",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "631f72d1de559130376df1fb"
      },
      "deploymentId" : 240000023
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000013
    },
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000005
    }
  ]
}
}
```

nrkumar93, 2023-05-23:
I am also facing this since yesterday morning. How can this happen arbitrarily on my local version? I am totally confused!

thowell, 2023-05-23:
Thanks for opening this issue. This is a known problem due to a change to the latest version of MuJoCo. MJPC currently pulls the latest MuJoCo version. 

A temporary fix is to change one line of code in the build dependencies: 
`build/_deps/mujoco-src/src/engine/engine_vis_init.c`
set line 93 to: 
`{"Mesh Tree",       "0", "!"}`

erez-tom, 2023-05-23:
This bug is the result of an [upstream change](https://github.com/deepmind/mujoco/commit/a09246c095471df1c60b68478e4e6805f17e5837) to the MuJoCo codebase.  Since cmake pulls the latest version of MuJoCo, such bugs can appear without changes to downstream code.
As for this specific bug, see fix [here](https://github.com/deepmind/mujoco_mpc/commit/dd830448a11c4699d84da58a0c7ad4489f1039fc).  We will [merge](https://github.com/deepmind/mujoco_mpc/pull/113) this to the main branch ASAP.

closed 2023-05-23 by erez-tom

poppy-alpaca, 2023-05-24:
Thanks for the fast triage!