google-deepmind/mujoco_mpc #114: Mujoco plugin is not working in mjpc

ssk-yoshimura, 2023-05-24:
I'd like to use self-defined mujoco plugins in mjpc, however, those plugins work well in mujoco, but produce error when executing mjpc. 

The problem can be reproduced by adding the default mujoco plugin:
`
<extension>
  <plugin plugin="mujoco.sensor.touch_grid"/>
</extension>
`
into mjpc/tasks/manipulation/task_panda_bring.xml.

After rebuilding and executing ./mjpc, the error comes as the following:
[image]

I'm using the latest mujoco_mpc and the version of fetched mujoco has been changed to 2.3.5.

Any advice on how to solve this problem? Look forward to your response!

saran-t, 2023-05-24:
You need to register the plugin first. There are two ways you could do this, either `dlopen` the sensor library in your code (you can just grab it from our prebuilt package), or link against it with `-l` at build time.

erez-tom, 2023-05-24:
How have you incorporated your code into the build process? Did you add it as a dependency of libmjpc, or the mjpc executable?  Do the build logs show that your code was indeed built? Does your plug-in binary end up where MuJoCo expects to load it? Is it being loaded successfully, and then not found?
I'll keep this open for now. Let us know what ends up working for you!

ssk-yoshimura, 2023-05-24:
I added `dlopen("my_mujoco_library.so", RTLD_LAZY)`  to mjpc/tasks/main.cc, and it works!

Also, I added the library to `target_link_libraries` in mjpc/CMakeList.txt, then it works without dlopen.

Thank you!