google-deepmind/mujoco_mpc #116: support more robots

GilgameshD, 2023-05-27:
Hi, thanks for building this amazing project! I wonder if there is any plan to support more robots, for example, Unitree go1, which should be very similar to A1. Thanks

thowell, 2023-05-27:
Thanks for your interest in MJPC. If you would like to add another task, take a look at this [part](https://github.com/deepmind/mujoco_mpc/blob/main/docs/OVERVIEW.md#task-specification) of the documentation. 

It's pretty straight forward to modify the A1 task by swapping in the Go1 xml model (and the related assets) from [Menagerie](https://github.com/deepmind/mujoco_menagerie/tree/main/unitree_go1).

@kevinzakka do you have the Go1 task floating around somewhere?

kevinzakka, 2023-05-28:
Go1 XML is now in Menagerie — it should be super straightforward to replicate the A1 task for it, mostly just a matter of copying over the XML and assets and adding some sites.

GilgameshD, 2023-05-29:
Thanks for replying. I tried to replace A1.xml with GO1.xml and fix all compatibility problems. I can successfully start the UI with GO1 now. But the robot always flips and cannot walk. I guess there are still some parameters in the task file that are not compatible with GO1. Is there anything I should change in [this file](https://github.com/deepmind/mujoco_mpc/blob/main/mjpc/tasks/quadruped/quadruped.cc)?

thowell, 2023-05-29:
You will likely need to change some of the parameters [here](https://github.com/deepmind/mujoco_mpc/blob/47e54222009bb454d22952c0a79d9a83c6659f01/mjpc/tasks/quadruped/quadruped.h#L121) like `kHeightQuadruped`.

GilgameshD, 2023-05-29:
will try, thanks!

lyuyiqi, 2023-06-13:
Hi all. I am working on a similar project and I've had GO1 working on my simulator. However, everytime I turned off "action" the robot always straightened its feet and jump off the ground (restore to its initial abduction, hip and knee angles) instead of lying down due to loss of actuation like original A1 model does. I would really appreciate it if you could give me some hint to solve this problem.

kevinzakka, 2023-06-13:
@lyuyiqi Are you using torque or position actuators?

lyuyiqi, 2023-06-13:
Thank you for your reply. I just checked the model again and indeed the actuators are position type. After I replace them with general type and now it works very well! Thank you again for your instruction!

yuvaltassa, 2023-06-21:
Did you try reducing the timestep?

fucker007, 2023-06-22:
Thanks for replying，It is work well when setting a smaller timestep.  I am learning the code of humanoid walker, but I am wondering why -7 when setting posture.   
code path: ` mujoco_mpc/tree/main/mjpc/tasks/humanoid/walk/walk.cc,  line 122. `  
```
  // ----- posture ----- //  
  mju_copy(&residual[counter], data->qpos + 7, model->nq - 7);  
  counter += model->nq - 7;  
```
@yuvaltassa thanks for your replying, best wish.

kevinzakka, 2023-06-22:
-7 because we don't want to index the freejoint (3 dims for pos, 4 for quat).

fucker007, 2023-06-23:
> -7 because we don't want to index the freejoint (3 dims for pos, 4 for quat).

Thanks for replying，This makes me understand correctly, bless you.

fucker007, 2023-07-17:
I have tried some tricks on cassie , however  it's diffcult to walk by mujoco_mpc,  so  is it possible ? any suggestion, be grateful.

thowell, 2023-12-22:
See #206 for using cassie model with MJPC.

closed 2023-12-22 by thowell