google-deepmind/mujoco_mpc #124: Customized button in the Agent section

lyuyiqi, 2023-06-15:
Dear deepmind team,

I am a student working to collect some data of my robot on mujoco_mpc. Now I wish to add two buttons to start and stop my data collection in the Agent section of the UI. I tried to add some status flags in the header, and then in the source file, I defined the default values of them and add two lines in the UI defining function. However, when I tried to run the simulator it just got frozen and couldn't even load the model. Could you please give me some instruction to fix this?

Here are what I added:

---
agent.h:
```
// status flags, logically should be bool, but mjUI needs int pointers
int plan_enabled;
int action_enabled;
int visualize_enabled;
int allocate_enabled;
int plot_enabled;
int gui_task_id = 0;
//--------------------What I added--------------------
int record_enabled;
int save_enabled;
//--------------------What I added--------------------
```
---
agent.cc:
```
// status
plan_enabled = false;
action_enabled = true;
visualize_enabled = false;
allocate_enabled = true;
plot_enabled = true;
//--------------------What I added--------------------
record_enabled = false;
save_enabled = false;
//--------------------What I added--------------------

// ----- agent ----- //
mjuiDef defAgent[] = {
{mjITEM_SECTION, "Agent", 1, nullptr, "AP"},
{mjITEM_BUTTON, "Reset", 2, nullptr, " https://github.com/deepmind/mujoco/issues/459"},
{mjITEM_SELECT, "Planner", 2, &planner_, ""},
{mjITEM_CHECKINT, "Plan", 2, &plan_enabled, ""},
{mjITEM_CHECKINT, "Action", 2, &action_enabled, ""},
{mjITEM_CHECKINT, "Plots", 2, &plot_enabled, ""},
{mjITEM_CHECKINT, "Traces", 2, &visualize_enabled, ""},
//--------------------What I added--------------------
{mjITEM_CHECKINT, "Record", 2, &record_enabled, ""},
{mjITEM_CHECKINT, "Save", 2, &save_enabled, ""},
//--------------------What I added--------------------
{mjITEM_SEPARATOR, "Agent Settings", 1},
{mjITEM_SLIDERNUM, "Horizon", 2, &horizon_, "0 1"},
{mjITEM_SLIDERNUM, "Timestep", 2, &timestep_, "0 1"},
{mjITEM_SELECT, "Integrator", 2, &integrator_, "Euler\nRK4\nImplicit\nFastImplicit"},
{mjITEM_SEPARATOR, "Planner Settings", 1},
{mjITEM_END}};
```

thowell, 2023-06-15:
The ordering of the mjITEMs matters here since some of the items are later modified, see [agent.cc](https://github.com/thowell/mujoco_mpc/blob/01665a8d9de0ab5aa8aefb20a150efab8e3ce160/mjpc/agent.cc#L626).

If you want to add additional GUI elements to the agent tab I would recommend doing this after "Integrator" and before "Planner Settings".

lyuyiqi, 2023-06-15:
Resolved with your advised. Thank you a lot!

closed 2023-06-15 by lyuyiqi