google-deepmind/mujoco_mpc #125: Build Problems!

warmmm, 2023-06-15:
Hi,I am tring to learn this program in my computer, but it was failed when I build it in my VS code,the problems may be these:
‘void mjpc::{anonymous}::TestHull(int, mjtNum*, int, int*)’ defined but not used [-Werror=unused-function]
[{
	"resource": "/home/chen/mujoco_mpc/mjpc/test/utilities/utilities_test.cc",
	"owner": "cmake-build-diags",
	"severity": 8,
	"message": "taking address of temporary array",
	"source": "GCC",
	"startLineNumber": 89,
	"startColumn": 36,
	"endLineNumber": 89,
	"endColumn": 1000
}]
[{
	"resource": "/home/chen/mujoco_mpc/mjpc/test/utilities/utilities_test.cc",
	"owner": "cmake-build-diags",
	"severity": 8,
	"message": "taking address of temporary array",
	"source": "GCC",
	"startLineNumber": 99,
	"startColumn": 36,
	"endLineNumber": 99,
	"endColumn": 1000
}]

and the other 7 problems below:
[image]

I would be appreciated for you if you could help me!!!

lyuyiqi, 2023-06-15:
I never had a good luck with GCC but successfully made it with clang.

thowell, 2023-06-15:
We are using clang, please see: [Build Issues](https://github.com/deepmind/mujoco_mpc#build-issues).

fucker007, 2023-06-17:
I get  same problem, I fix this as follow:   
```  
sudo bash -c "$(wget -O - https://apt.llvm.org/llvm.sh)"  
```  
then    
```  
mkdir build  
cd build  
cmake .. -DCMAKE_C_COMPILER=clang-16 -DCMAKE_CXX_COMPILER=clang++-16   
```

warmmm, 2023-06-21:
> We are using clang, please see: [Build Issues](https://github.com/deepmind/mujoco_mpc#build-issues).

I try it again and fellow .github/workflows/build.yml ,but when I test , the problem become: No tests were found!!!

so, what should I solve this issue?

thowell, 2023-08-02:
The tests run for me if I call `ctest` from `mujoco_mpc/build/mjpc/test`.

@nimrod-gileadi is there a way for us to modify the top-level CMakeList so that the VSCode tools can automatically find the tests? I think we need to do something like [this](https://stackoverflow.com/questions/64099136/visual-studio-2019-test-explorer-did-not-find-c-google-tests), but I couldn't get their solution to work by just adding `enable_testing()`.

nimrod-gileadi, 2023-08-02:
> > We are using clang, please see: [Build Issues](https://github.com/deepmind/mujoco_mpc#build-issues).
> 
> I try it again and fellow .github/workflows/build.yml ,but when I test , the problem become: No tests were found!!!
> 
> so, what should I solve this issue?

What if you run the same command we use in GitHub Actions:

```
ctest -C Release --output-on-failure .
```

amdee, 2023-09-29:
This works for me, I am using Ubuntu 22.04, Clang version 14, vscode, and cmake vscode extension by Microsoft

1. Clone the repo to your local machine and open it with vscode
```bash
git clone https://github.com/google-deepmind/mujoco_mpc.git
cd mujoco_mpc && code .
```
2. Change the cmake build variant to Release by shift + ctr + p and select Release or change in picture below
[image]

closed 2024-01-12 by thowell