google-deepmind/mujoco_mpc #132: Perturbation

scakki, 2023-07-13:
Hello,
How can I set perturbation for the Quadreped A1 robot?

Thank you

scakki, 2023-07-13:
Is there any other way? Like using Perturb API and the XFRC_APPLIED?

thowell, 2023-07-14:
Forces and torques can be applied to a specific body in the GUI using the click-and-drag interface, please see the [docs](https://github.com/deepmind/mujoco_mpc/blob/main/docs/GUI.md#user-guide).

thowell, 2023-07-15:
Currently, there is not a way to do this from the GUI. You could modify the line before mj_step [here](https://github.com/deepmind/mujoco_mpc/blob/28bc73029c214dfae8ca3917f814adaf22b35cc4/mjpc/app.cc#L258) or [here](https://github.com/deepmind/mujoco_mpc/blob/28bc73029c214dfae8ca3917f814adaf22b35cc4/mjpc/app.cc#L281) to set `xfrc_applied` at certain times. 

Alternatively, you can use the [Python interface](https://github.com/deepmind/mujoco_mpc/blob/main/docs/OVERVIEW.md#python-api) to set `xfrc_applied`.

scakki, 2024-07-01:
Hello Thowell,

Thank you for your previous response and suggestions. I have been using the click-and-drag option as mentioned, which has worked well for my needs so far.

Recently, I attempted to implement the code changes you suggested to set xfrc_applied before mj_step. However, I encountered some difficulties and am unable to get it to work as expected. Below is the test code I used. I can verify that the force is being applied after 5 seconds (as the value is being printed), but I do not notice any difference in the simulation:

```
            if (d->time >= 5) { 
                d->xfrc_applied[6 * 3 + 1] = 1000;
                std::cout << d->xfrc_applied[6 * 3 + 1] << std::endl;
            }
```

Could you please provide some additional guidance or an example on how to correctly implement this? Any help would be greatly appreciated.

Thank you for your assistance!

thowell, 2024-07-09:
Where are you setting `xfrc_applied` in the code? Can you please provide a link to your fork?

thowell, 2024-07-11:
@scakki in app.cc set `xfrc_applied` [here](https://github.com/google-deepmind/mujoco_mpc/blob/1f78168912b4fe4af30365198c9f4e3ecf025ba1/mjpc/app.cc#L327) and [here](https://github.com/google-deepmind/mujoco_mpc/blob/1f78168912b4fe4af30365198c9f4e3ecf025ba1/mjpc/app.cc#L353).

scakki, 2024-07-11:
Thank you very much for your response. Perturbation is now working.

closed 2024-07-11 by scakki

ppap36, 2025-04-22:
Hello!

I am new to mjpc, I verified your answer but  can you tell me why the `xfrc_applied` must be put at here?
Why I cannot modify it in my own task file?

Looking forward to your reply.