google-deepmind/mujoco_mpc #135: Examples in the mjpc GUI

emailweixu, 2023-07-15:
I am playing with the examples in mjpc GUI. Unfortunately, except cartpole, I couldn't see any meaningful behavior.
For example,  the Quadraped Fl robot seems only doing some random movement on the ground and is not able to stand up with the default setting or with changes I have tried.
For humanoid st, the robot could standup once a while but then fell down soon.

mjpc is compiled following the instruction in readme using vscode with clang-14. mujoco version is 2.3.6.

thowell, 2023-07-15:
Make sure you set CMake to 'Release' in VSCode when you build. How many threads does your machine have?

emailweixu, 2023-07-17:
> Make sure you set CMake to 'Release' in VSCode when you build. How many threads does your machine have?

Yes, it's compiled in "Release" mode and there are 32 threads in my machine.

thowell, 2023-07-19:
Reach out to me via email so we can get on a video call and I can help you get MJPC working on your machine.

emailweixu, 2023-07-20:
Here is a summary about the cause and the solution of the problem in case it helps others. Basically, the simulation in mjpc runs asynchronously with the planner. If the planner is not fast enough, it cannot finish the computation in time. In my case, the slowness is probably caused by the use of remote desktop. Giving the planner more time to react by slowing down the simulation speed (by pressing '-' key) solves the problem.

closed 2023-07-20 by emailweixu

yunsooSeo, 2025-08-05:
Hi, I believe I'm facing a similar issue to the one discussed here.

When I run the provided example tasks like humanoid_stand or humanoid_walk, the robots seem to behave reasonably well—even with the default timestep = 0.015—as long as I tune some parameters like balance. I don’t need to reduce the simulation speed to the minimum in those cases.

However, I’m now trying to run a similar setup using my own G1 model, using the same residual and transition structure as in the humanoid tasks. Initially, I was getting rollout divergence errors with very high qacc, so I reduced the planner timestep to 0.001. That eliminated the divergence errors, but now the robot still falls down quickly even with the simulation speed reduced to the minimum.

My question is: how can I tell whether this issue is due to a mismatch between the planner and simulation (e.g. planner being too slow to keep up), or whether it's something wrong with my residual definition or task configuration?

Any suggestions for debugging or confirming the source of the issue would be greatly appreciated. Thanks!