google-deepmind/mujoco_mpc #137: Is there a way to define new modes?

Jasperora, 2023-07-23:
Take quadruped for example, there are five modes. They are quadruped, biped, walk, scramble, flip, respectively.
I wonder if there is something like an API such that given reference motion trajectories, then the motion can be visualized in GUI.
So that we can define new motion beyond the given five modes.
Thanks.

yuvaltassa, 2023-07-23:
Did you look at the Humanoid Track task? I think @hartikainen has a better version of this task actually. Kristian, did you ever get around to updating it?

thowell, 2023-07-23:
If your question is about adding a new button to the GUI, you can specify this in the xml using `task_transition` see this [example](https://github.com/deepmind/mujoco_mpc/blob/28bc73029c214dfae8ca3917f814adaf22b35cc4/mjpc/tasks/humanoid/tracking/task.xml#L7C1-L7C135).

If your question is about visualizing a behavior in the GUI that was designed offline and encoded as keyframes in the xml, this functionality is not currently available.

Jasperora, 2023-07-24:
Thanks for replying. I'm looking at the quadruped tasks.
What I want to do is to visualize a reference behavior that was designed offline and encoded as keyframes in the xml in GUI.

closed 2023-07-24 by Jasperora