google-deepmind/mujoco_mpc #138: Tutorial or Document Needed!

SiyuanHuang95, 2023-07-24:
Hi,

thanks for your contribution and open-sourcing! I have watched some demo videos from yours and quite impressive!

I am with the background of SLAM / 3D point cloud perception, and would like to have some trials for the combination of dynamics like MJPC with other stuff. However, I found that there is almost no docs for this project,

Can you provide any docs or information wrt the usage. Basically, I am trying to control a quadruped's pose with python API and it's pose could be online solved with MJPC.

Bests

thowell, 2023-07-24:
Documentation is available [here](https://github.com/deepmind/mujoco_mpc/tree/main/docs).

SiyuanHuang95, 2023-07-25:
Thanks for your quick reply.

The link you shared is more about the GUI introduction, would there be any possibility for the Python API Doc or some demo codes?

closed 2023-07-28 by SiyuanHuang95