google-deepmind/mujoco_mpc #140: Python install problem

SiyuanHuang95, 2023-07-25:
Hi,

thanks for the sharing.

After cmake and make, I could run the GUI with C++ version successfully, very impressive!

And then, I removed the build/ folder and  tried to use 

```
cd python/
pip install -e .
```

to install the python version, however, the process has already taken about 2 hrs, and I can only see something like " Running setup.py develop for mujoco-mpc" in the terminal, at the same time, I found that the build/ folder is growing consistently:
```
Every 5.0s: du --max-depth=1 -h                                                                 

1.5G    ./_deps
16K     ./mjpc
400K    ./CMakeFiles
8.0K    ./grpc
1.5G    .
````

Was that normal? Or is there anything wrong with my operation?

SiyuanHuang95, 2023-07-26:
After some hours waiting, it gives me:

```
Successfully installed mujoco-mpc-0.1.0
```

with the build folder:
```
288K    ./gens
615M    ./lib
520K    ./protos
3.1G    ./_deps
4.0K    ./http_archives
19M     ./mjpc
311M    ./bin
1.6M    ./CMakeFiles
16M     ./grpc
4.0G    .
```

however, when try 

```
python ui_agent_test.py
```

error raises:

```
ModuleNotFoundError: No module named 'mujoco_mpc.proto'
```

Should I re-install it?

SiyuanHuang95, 2023-07-26:
After many trail and errors, I finally successfully install it. Some lessons:

1. Do not use pip install -e, just use install, this will speed-up the process if you do not need to modify some source codes.
2. USE python>=3.10!

SiyuanHuang95, 2023-07-26:
I have successfully run some test scripts:

1. python agent_test.py -> success
2. python ui_agent_test.py -> success
3. python estimator_test.py -> cannot import name 'estimator_pb2' from 'mujoco_mpc.proto' (/home/XXX/anaconda3/envs/mjpc/lib/python3.10/site-packages/mujoco_mpc/proto/__init__.py)

I install the packages with:
```
python setup.py install
python setup_estimator.py install
```

and both give me the success flag.

SiyuanHuang95, 2023-07-28:
I also tried with 

```
pip install PATH/mujoco_mpc/python
```

and the terminal gives me:

```
**byte-compiling build/bdist.linux-x86_64/egg/mujoco_mpc/proto/estimator_pb2.py to estimator_pb2.cpython-310.pyc**
**byte-compiling build/bdist.linux-x86_64/egg/mujoco_mpc/proto/agent_pb2.py to agent_pb2.cpython-310.pyc**
**byte-compiling build/bdist.linux-x86_64/egg/mujoco_mpc/proto/estimator_pb2_grpc.py to estimator_pb2_grpc.cpython-310.pyc**
...

Using /XX/anaconda3/envs/mjpc/lib/python3.10/site-packages
Finished processing dependencies for mujoco-mpc==0.1.0
```

So I thought the install process is successful?

But the python estimator_test.py -> cannot import name 'estimator_pb2' from 'mujoco_mpc.proto' (/home/XXX/anaconda3/envs/mjpc/lib/python3.10/site-packages/mujoco_mpc/proto/init.py)

Any help and information is indeed appreciate!

thowell, 2023-08-02:
I do the following:
change directories to `python/`
then run `python setup.py install`

thowell, 2023-08-02:
If you still encounter issues, feel free to reach out via email to me and I can walk you through the install and using MJPC python in a colab notebook.

thowell, 2023-08-02:
#142 should fix this

thowell, 2023-08-02:
Are you installing the estimator first: `python python/setup_estimator.py install`?

SiyuanHuang95, 2023-08-02:
> Are you installing the estimator first: `python python/setup_estimator.py install`?

Nopes, I directly run with 

```
pip install PATH/mujoco_mpc/python
```

will this make a big difference?

Hither1, 2023-09-17:
Hi, is there any required version for python or other packages in order to run `python python/setup.py install` successfully?

Hither1, 2023-09-28:
Is setup_estimator.py replaced by setup_batch.py in the latest version?

thowell, 2023-12-22:
@SiyuanHuang95 the README is updated with improved installation instructions.
@Hither1 the latest filename is setup_direct.py.

closed 2023-12-22 by thowell