google-deepmind/mujoco_mpc #157: Is there a way to make this work wtih 'position

alecGraves, 2023-08-20:
<position

thowell, 2023-08-20:
Position actuators work, see the [hand task](https://github.com/deepmind/mujoco_mpc/blob/40cf509e8906273f0f69275f47f5f5f6561e642d/mjpc/tasks/hand/right_hand.xml#L324). You may need to modify the cost function to work with different actuators. A common change is to penalize `actuator_force` instead of `ctrl` ([example](https://github.com/deepmind/mujoco_mpc/blob/40cf509e8906273f0f69275f47f5f5f6561e642d/mjpc/tasks/quadruped/quadruped.cc#L159)).

Do you have an example task we can look at?

alecGraves, 2023-08-20:
Ok, using actuator_force might be it. Thanks for pointing me to the hand task; I will take a closer look at it.

Here is [my task (gitlab)](https://gitlab.com/RoadrunnerDynamics/mujoco_mpc/-/blob/dam-leg/mjpc/tasks/dam_leg) (`<actuator>` defined at the bottom of leg.xml, original position actuators commented out). I just copied the cartpole task and threw a custom model in it to see if I could get it to move around.

I modified the residuals to penalize velocity difference from 1.0, just to get the model to move. When I use position actuators, no motion occurs despite an elevated cost. The optimization  improvement graph is stuck at zero. 

With motor actuators, the leg flails around as expected. Maybe my model's constraints are interacting with the optimizer in a funny way when I use position actuators.

Video of movement with motor actuators: https://youtu.be/wAMkcH5_GVk

thowell, 2023-08-20:
Take a look at this [branch](https://github.com/thowell/mujoco_mpc/tree/dam_leg).

I changed the position actuators to use `ctrllimited` and modified the way the residual is computed.

You can probably take it from here. Let us know if you encounter any additional blockers..

alecGraves, 2023-08-20:
Thanks for the example!

Based on your example and some more testing, it looks like adding ctrllimited and explicit ranges allows planning and motion as expected with `<position>` actuators.

Also, thanks for demonstrating how to include `data->actuator_force`.

closed 2023-08-20 by alecGraves