google-deepmind/mujoco_mpc #184: Mujoco update breaks build.

cybertale, 2023-09-14:
Mujoco's latest update 86d8b91 removes 'solver_iter' and change its calling logic, which breaks mujoco_mpc's building.
I fixed this build problem with reverting latest mujoco commits. I think we should specify commit version of mujoco when pulling the repo.

erez-tom, 2023-09-14:
I'm curious: which part of `mujoco_mpc`'s build was broken by the change?  I don't see `solver_iter` mentioned in the codebase, but I'm guessing maybe it's implicit in the Python build of MuJoCo?

erez-tom, 2023-09-14:
Would the issue have been avoided if you used the [Python dev version](https://pypi.org/project/mujoco-dev/)?

yuvaltassa, 2023-09-14:
@erez-tom search [here](https://github.com/google-deepmind/mujoco_mpc/commit/ab5edf26361323c8d4de4ee61ae40ddcca61b66c#diff-db90f64a7aa728d7a52e70246a07ab3f928b1c8359eeea374c6a9dca269aff21L329) for solver_iter and you'll find it

thowell, 2023-09-29:
I want to suggest that **we build MJPC with specific MuJoCo versions** as well. 

I understand that internally it makes sense to always use the latest version of MuJoCo. However, externally it's very problematic for MJPC to build with the latest MuJoCo. This has been a recurring issue that MuJoCo changes break MJPC. Additionally, without a specific MuJoCo version in the MJPC commit history this package is not easily backward compatible.

Because deepmind branch is infrequently synced to main, it should be easy to specify the latest public version of MuJoCo that works with changes that will be merged from deepmind branch.

@saran-t @nimrod-gileadi

closed 2023-12-22 by thowell