google-deepmind/mujoco_mpc #185: Mujoco commit 1afd98e breaks the build

nkhosh, 2023-09-15:
I get the following cmake error since [this commit](https://github.com/google-deepmind/mujoco/commit/1afd98ec4b7339b3bdace4f51810c2ce160ab978) on mujoco:

`[cmake] FAILED: sdflib-populate-prefix/src/sdflib-populate-stamp/sdflib-populate-patch ~/mujoco_mpc/build/_deps/sdflib-subbuild/sdflib-populate-prefix/src/sdflib-populate-stamp/sdflib-populate-patch 
[cmake] cd ~/mujoco_mpc/build/_deps/sdflib-src && git apply --reject --whitespace=fix ~/mujoco_mpc/cmake/sdflib-optional-dependencies.patch && /usr/bin/cmake -E touch ~/mujoco_mpc/build/_deps/sdflib-subbuild/sdflib-populate-prefix/src/sdflib-populate-stamp/sdflib-populate-patch`
`[cmake] error: can't open patch '~/mujoco_mpc/cmake/sdflib-optional-dependencies.patch': No such file or directory`
`[cmake] ninja: build stopped: subcommand failed.`

I reverted back to the previous commit on my side so I'm not blocked by this, just reporting.

OSKOOO, 2023-09-16:
I ran into the same issue. It seems like the path generated by ${_ARGS_PATCH_COMMAND} in mujoco_mpc/cmake/FindOrFetch.cmake is incorrect. I fixed it manually by navigating to FindOrFetch.cmake and then after line 114 I use:
 
      if("${_ARGS_LIBRARY_NAME}" STREQUAL "sdflib")
        set(_ARGS_PATCH_COMMAND "git;apply;--reject;--whitespace=fix;${CMAKE_SOURCE_DIR}/build/_deps/mujoco-src/cmake/sdflib-optional-dependencies.patch")
        endif()

quagla, 2023-09-16:
This is now fixed. Thanks for reporting.

closed 2023-09-16 by quagla