google-deepmind/mujoco_mpc #194: Can't build in Windows

NghiaPhamttk27, 2023-09-30:
I am using Windows 11. I have successfully configured it using CMake. However, when I try to build, it gives me a warning, and I can't run it anymore. I am using GCC (Rev7, Built by MSYS2 project) 13.1.0. This is my terminal
```
15: error: too many arguments for format [-Werror=format-extra-args]
 1285 |     mju_error("mj_stackAlloc: insufficient memory: max = %zu, available = %zu, requested = %zu "
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
[19/404   4% :: 3.117] Building CXX object _deps/mujoco-build/CMakeFiles/lodepng.dir/__/lodepng-src/lodepng.cpp.obj
[19/404   4% :: 12.022] cmd.exe /C "cd /D C:\Users\PC\Code\mujoco_mpc-main\mujoco_mpc-main\build && C:\msys64\mingw64\bin\ninja.exe -C C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build -f build.ninja"
FAILED: CMakeFiles/MyTarget C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/CMakeFiles/MyTarget 
cmd.exe /C "cd /D C:\Users\PC\Code\mujoco_mpc-main\mujoco_mpc-main\build && C:\msys64\mingw64\bin\ninja.exe -C C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build -f build.ninja"
ninja: Entering directory `C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build'
[1/2   0% :: 0.025] Re-checking globbed directories...
[18/404   0% :: 0.945] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_callback.c.obj
[19/404   0% :: 0.989] Building RC object _deps/mujoco-build/CMakeFiles/mujoco.dir/dist/mujoco.rc.obj
FAILED: _deps/mujoco-build/CMakeFiles/mujoco.dir/dist/mujoco.rc.obj 
C:\msys64\ucrt64\bin\windres.exe -O coff -DCCD_STATIC_DEFINE -DMC_IMPLEM_ENABLE -DMUJOCO_DLL_EXPORTS -DMUJOCO_TINYOBJLOADER_IMPL -DTINYXML2_DEBUG -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GNU_SOURCE -DmjUSEPLATFORMSIMD -Dmujoco_EXPORTS -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/marchingcubecpp-src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/include -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/ccd-src/src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/ccd-build/src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/lodepng-src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/qhull-src/src/libqhull_r -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/tinyobjloader-src -I C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/tinyxml2-src  C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/dist/mujoco.rc _deps/mujoco-build/CMakeFiles/mujoco.dir/dist/mujoco.rc.obj
C:\msys64\ucrt64\bin\windres.exe: C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/dist/mujoco.rc:23: syntax error
[19/404   0% :: 1.095] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_primitive.c.obj
[19/404   0% :: 1.160] Building CXX object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_crossplatform.cc.obj
[19/404   1% :: 1.172] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_convex.c.obj
[19/404   1% :: 1.173] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_box.c.obj
[19/404   1% :: 1.250] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_sdf.c.obj
[19/404   1% :: 1.309] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_driver.c.obj
[19/404   2% :: 1.330] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_island.c.obj
[19/404   2% :: 1.379] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_derivative_fd.c.obj
[19/404   2% :: 1.445] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_forward.c.obj
[19/404   2% :: 1.522] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_inverse.c.obj
[19/404   3% :: 1.543] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_core_smooth.c.obj
[19/404   3% :: 1.561] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_passive.c.obj
[19/404   3% :: 1.565] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_core_constraint.c.obj
[19/404   3% :: 1.628] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_derivative.c.obj
[19/404   4% :: 2.979] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_io.c.obj
FAILED: _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_io.c.obj 
C:\msys64\ucrt64\bin\cc.exe -DCCD_STATIC_DEFINE -DMC_IMPLEM_ENABLE -DMUJOCO_DLL_EXPORTS -DMUJOCO_TINYOBJLOADER_IMPL -DTINYXML2_DEBUG -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GNU_SOURCE -DmjUSEPLATFORMSIMD -Dmujoco_EXPORTS -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/marchingcubecpp-src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/include -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/ccd-src/src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/ccd-build/src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/lodepng-src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/qhull-src/src/libqhull_r -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/tinyobjloader-src -IC:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/tinyxml2-src -g -std=c11 -fvisibility=hidden -fdata-sections -ffunction-sections -mavx -mpclmul -Werror -Wall -Wpedantic -Wimplicit-fallthrough -Wunused -Wvla -Wno-int-in-bool-context -Wno-sign-compare -Wno-unknown-pragmas -Wimplicit-fallthrough=5 -Wno-maybe-uninitialized -MD -MT _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_io.c.obj -MF _deps\mujoco-build\CMakeFiles\mujoco.dir\src\engine\engine_io.c.obj.d -o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_io.c.obj -c C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c: In function 'stackallocinternal':
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c:1285:59: error: unknown conversion type character 'z' in format [-Werror=format=]
 1285 |     mju_error("mj_stackAlloc: insufficient memory: max = %zu, available = %zu, requested = %zu "
      |                                                           ^
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c:1285:76: error: unknown conversion type character 'z' in format [-Werror=format=]
 1285 |     mju_error("mj_stackAlloc: insufficient memory: max = %zu, available = %zu, requested = %zu "
      |                                                                            ^
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c:1285:93: error: unknown conversion type character 'z' in format [-Werror=format=]
 1285 |     mju_error("mj_stackAlloc: insufficient memory: max = %zu, available = %zu, requested = %zu "
      |                                                                                             ^
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c:1285:15: error: format '%d' expects argument of type 'int', but argument 2 has type 'uintptr_t' {aka 'long long unsigned int'} [-Werror=format=]
 1285 |     mju_error("mj_stackAlloc: insufficient memory: max = %zu, available = %zu, requested = %zu "
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1286 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
 1287 |               stack_info->bottom - stack_info->limit, stack_available_bytes, stack_required_bytes,
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                  |
      |                                  uintptr_t {aka long long unsigned int}
C:/Users/PC/Code/mujoco_mpc-main/mujoco_mpc-main/build/_deps/mujoco-src/src/engine/engine_io.c:1286:23: note: format string is defined here
 1286 |               "(ne = %d, nf = %d, nefc = %d, ncon = %d)",
[... cut ...]
[22/398   5% :: 8.871] Building C object _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_print.c.obj
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
build finished with error(s).

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

saran-t, 2023-09-30:
What version of GCC are you using?

saran-t, 2023-10-02:
To clarify, our we regularly build and test against GCC and Clang on Linux, and MSVC and Clang on Windows, and whatever is happening here isn't reproducible in our CI setup. I suspect that the issue lies in the specific toolchain that you're using (mingw through msys2 in this case).

NghiaPhamttk27, 2023-10-02:
Thank you for the response, I didn't use MSYS2 and everything was fine when I configured and built it in Visual Studio 2022. However, I'm facing an issue with display, probably related to my AMD's chip and GPU in my laptop. I found this issue on Mujoco (https://github.com/google-deepmind/mujoco/issues/894) but still haven't been able to resolve it, and it seems Mujoco MPC is also affected.
[image]

yuvaltassa, 2023-10-03:
Yes, we are aware of the AMD render issue. We haven't had an AMD GPU on hand to be able to reproduce but will soon have one. Follow the issue you referenced above for updates.

closed 2023-10-03 by yuvaltassa