google-deepmind/mujoco_mpc #206: QAcc Divergence when importing Cassie model to mjpc

manuaatitya, 2023-10-24:
Hi, I'm trying to import the Cassie Model, [folder link](https://drive.google.com/drive/folders/1VFkNFE5yowtSpfLabOw8lVPMj-UwsD0c?usp=share_link), (the model is same as from [cassie-mujoco-sim](https://github.com/osudrl/cassie-mujoco-sim/tree/master/model) just with the sensor tags rearranged for the purpose of residuals). 

One of the main issues is I get the divergence of QAcc in some of the joints.
[image]

The same model works well with just mujoco and I couldn't reason out as to why this is happening in mjpc. 

Tried a bunch of things like changing the integrator and adding damping to the joints but doesn't seem to solve the issue. (The Cassie.cc file doesn't do much, just trying to import Cassie into this framework but haven't been successful) Any help to solve this would be appreciated !!!

Thanks in advance!

yuvaltassa, 2023-10-25:
The Cassie model is not super stable IIRC.

My main guess (haven't looked at your file) is that your MJPC task is defining a planner timestep that is too large

manuaatitya, 2023-10-26:
Thanks, it fixed the issue I had to set the tilmestep to 0.0001 and anything above that and the planner gives out rollout divergence error. Thanks for the help. I played around with the time steps but didn't realize I had to keep it very low for it to work.

closed 2023-12-22 by thowell