google-deepmind/mujoco_mpc #207: Tag and version mjpc

vikashplus, 2023-10-30:
Hi would it be possible to version and tag **mujoco_mpc**. Its under active development and being able to identify stable points in the repo has been a challenge. 

Also shout out to @thowell for always helping me navigating the changes.

thowell, 2023-12-22:
MJPC commits are now [pinned to a specific MuJoCo version ](https://github.com/google-deepmind/mujoco_mpc/blob/dc00a680ecd25175dad0ef8e5e6a4e1f1b22176c/CMakeLists.txt#L59).

vikashplus, 2023-12-22:
@thowell  -- any plans for tagging the MJPC itself?

erez-tom, 2023-12-26:
Since we don't release artefacts, why do you need us to do the tagging on our side?  Couldn't you just grab the hash of a snapshot that's proving useful to you and stick with it?

vikashplus, 2023-12-26:
Tags are useful to identify where are the stable points, especially for repos that are under continuous development. Not all hashes are stable. Without tags, one has to search all the hash commits to find a place where the repo was most stable. Developers adding tags to specific hashes can make life of a user easier as they indicate which tagged-hashes are stable points for users to resort to.

thowell, 2024-01-14:
I agree with @vikashplus, this is a really useful feature.

@erez-tom I think the latest branch is pretty stable. Can you tag a commit? Going forward perhaps we can tag a version of MJPC when there is a major MuJoCo version update (we would want to bump the MuJoCo version in MJPC anyway)?

nimrod-gileadi, 2024-01-15:
Done. https://github.com/google-deepmind/mujoco_mpc/releases/tag/v0.0.1