google-deepmind/mujoco_mpc #214: Build Issues with Clang-14

hongmindavidkim, 2023-11-06:
Hi, with the cmake tools and clang-14, I am getting this below error message when trying to build. 
I've tried installing `protobuf-compiler` and `libprotobuf-dev` which did not resolve the issue. 

I was able to build this, couple months ago with the same setup, but when I tried building again today, it gave me this error. 

Any help to resolve this would be greatly appreciated!!

```
[main] Configuring project: mujoco_mpc 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang-14 -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++-14 -S/home/david/dev/mujoco-google-deepmind/mujoco_mpc -B/home/david/dev/mujoco-google-deepmind/mujoco_mpc/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- mujoco::FindOrFetch: checking for targets in package `mujoco`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `mujoco` - target `mujoco::mujoco` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `mujoco`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `qhull`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `qhull` - target `qhull` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `qhull`
[cmake] -- 
[cmake] -- ========== qhull Build Information ==========
[cmake] -- Build Version:                             8.1-alpha1
[cmake] -- Install Prefix (CMAKE_INSTALL_PREFIX):     /usr/local
[cmake] -- Binary Directory (BIN_INSTALL_DIR):        bin
[cmake] -- Library Directory (LIB_INSTALL_DIR):       lib
[cmake] -- Include Directory (INCLUDE_INSTALL_DIR):   include
[cmake] -- Documentation Directory (DOC_INSTALL_DIR): share/doc/qhull
[cmake] -- Man Pages Directory (MAN_INSTALL_DIR):     share/man/man1
[cmake] -- CMake Directory (CMAKE_INSTALL_DIR):       lib/cmake/QHull
[cmake] -- PkgConfig Directory (PKGCONFIG_INSTALL_DIR):lib/pkgconfig
[cmake] -- Build Type (CMAKE_BUILD_TYPE):             Release
[cmake] -- Build static libraries:                    ON
[cmake] -- Build shared library:                      OFF
[cmake] -- Use shared library for linking apps:       OFF
[cmake] -- Build tests:				   OFF
[cmake] -- To override these options, add -D{OPTION_NAME}=... to the cmake command
[cmake] --   Build the debug targets                  -DCMAKE_BUILD_TYPE=Debug
[cmake] -- 
[cmake] -- To build and install qhull, enter "make" and "make install"
[cmake] -- To smoketest qhull, enter "ctest"
[cmake] -- 
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `qhull` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `tinyxml2`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `tinyxml2` - target `tinyxml2` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `tinyxml2`
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `tinyxml2` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `tinyobjloader`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `tinyobjloader` - target `tinyobjloader` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `tinyobjloader`
[cmake] CMake Deprecation Warning at build/_deps/tinyobjloader-src/CMakeLists.txt:5 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 2.8.12 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] 
[cmake] 
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `tinyobjloader` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `sdflib`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `sdflib` - target `SdfLib` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `sdflib`
[cmake] CMake Deprecation Warning at build/_deps/glm_lib-src/CMakeLists.txt:1 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 2.8.12 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] 
[cmake] 
[cmake] CMake Deprecation Warning at build/_deps/glm_lib-src/CMakeLists.txt:2 (cmake_policy):
[cmake]   Compatibility with CMake < 2.8.12 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] 
[cmake] 
[cmake] -- GLM is a header only library, no need to build it. Set the option GLM_TEST_ENABLE with ON to build and run the test bench
[cmake] -- Build spdlog: 1.9.2
[cmake] -- Build type: Release
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `sdflib` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `ccd`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `ccd` - target `ccd` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `ccd`
[cmake] CMake Deprecation Warning at build/_deps/ccd-src/CMakeLists.txt:1 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 2.8.12 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] 
[cmake] 
[cmake] CMake Warning (dev) at build/_deps/ccd-src/CMakeLists.txt:9 (project):
[cmake]   Policy CMP0048 is not set: project() command manages VERSION variables.
[cmake]   Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
[cmake]   command to set the policy and suppress this warning.
[cmake] 
[cmake]   The following variable(s) would be set to empty:
[cmake] 
[cmake]     PROJECT_VERSION
[cmake]     PROJECT_VERSION_MAJOR
[cmake]     PROJECT_VERSION_MINOR
[cmake]     PROJECT_VERSION_PATCH
[cmake] This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `ccd` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `mujoco`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `mujoco` - found
[cmake] -- mujoco::FindOrFetch: checking for targets in package `glfw3`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `glfw3` - target `glfw` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `glfw3`
[cmake] -- Using X11 for window creation
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `glfw3` - Done
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `mujoco` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `GTest`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `GTest` - target `gtest` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `googletest`
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `googletest` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `absl`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `absl` - target `absl::core_headers` not defined.
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp`
[cmake] -- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp` - Done
[cmake] -- mujoco::FindOrFetch: checking for targets in package `glfw`
[cmake] -- mujoco::FindOrFetch: checking for targets in package `glfw` - found
[cmake] -- Configuring done
[cmake] CMake Error at mjpc/CMakeLists.txt:35 (add_custom_command):
[cmake]   Error evaluating generator expression:
[cmake] 
[cmake]     $<TARGET_FILE:protoc>
[cmake] 
[cmake]   No target "protoc"
[cmake] 
[cmake] 
[cmake] CMake Error at mjpc/CMakeLists.txt:35 (add_custom_command):
[cmake]   Error evaluating generator expression:
[cmake] 
[cmake]     $<TARGET_FILE:protoc>
[cmake] 
[cmake]   No target "protoc"
[cmake] 
[cmake] 
[cmake] CMake Error at mjpc/CMakeLists.txt:35 (add_custom_command):
[cmake]   Error evaluating generator expression:
[cmake] 
[cmake]     $<TARGET_FILE:protoc>
[cmake] 
[cmake]   No target "protoc"
[cmake] 
[cmake] 
[cmake] -- Generating done
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
```

hartikainen, 2023-11-06:
I believe this is already being addressed by the Google folks but in the meanwhile you can fix the problem with the changes in #215. Let me know if for whatever reason those changes don't fix the issues on your side.

hongmindavidkim, 2023-11-07:
@hartikainen Hey, I was finally able to successfully build it on my Ubuntu 22.04! Below are the changes I made. 

First I made the changes referring to #215 and added the `--experimental_allow_proto3_optional` flag, which configured the project correctly but the build failed. 

I found that there was an issue using the `operator[]` on a `google::protobuf::Map` which expects a key of type `std::string` but it was passing a `std::string_view` type instead, which was introduced in C++17. 

So changing the lines of `(*task_parameters)[name].set_selection(...)` and `(*task_parameters)[name].set_numeric(...)` to `(*task_parameters)[std::string(name)].set_selection(...)` and `(*task_parameters)[std::string(name)].set_numeric(...)` fixed the problem and I was able to build successfully!

erez-tom, 2023-11-09:
This is great!  Do you want to send a PR with your changes?

hongmindavidkim, 2023-11-10:
@erez-tom Sure! I'm also trying to fix this on 20.04. I'll create a PR after I figure this one out.

thowell, 2023-12-22:
@hongmindavidkim is this issue resolved?

closed 2024-01-12 by thowell