google-deepmind/mujoco_mpc #218: Compiling with llvm@14 and ninja on MacOS 11 (x86_64) fails on linking minigzip with libz

keszegrobert, 2023-11-25:
This was my setup:
brew install llvm@14 
#it took a day or so

git clone https://github.com/google-deepmind/mujoco_mpc.git
cd mujoco_mpc
mkdir build
cd build
export CC=clang
export CXX=clang++
#edit cmake/MujocoLinkOptions.cmake to not build with --gc-sections
cmake .. -G Ninja -DMJPC_BUILD_GRPC_SERVICE:BOOL=ON
ninja
As I have seen at: https://github.com/google-deepmind/mujoco_mpc/blob/main/.github/workflows/build.yml

[2691/2754] Linking C executable bin/minigzip
FAILED: bin/minigzip 
: && /usr/local/opt/llvm@14/bin/clang -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk -mmacosx-version-min=11.7 -Wl,-search_paths_first -Wl,-headerpad_max_install_names  _deps/grpc-build/third_party/zlib/CMakeFiles/minigzip.dir/test/minigzip.o -o bin/minigzip  lib/libz.1.2.13.dylib && :
Undefined symbols for architecture x86_64:
  "_gzclose", referenced from:
      _gz_compress in minigzip.o
      _gz_uncompress in minigzip.o
  "_gzdopen", referenced from:
      _main in minigzip.o
  "_gzerror", referenced from:
      _gz_compress in minigzip.o
      _gz_uncompress in minigzip.o
  "_gzopen", referenced from:
      _file_compress in minigzip.o
      _file_uncompress in minigzip.o
      _main in minigzip.o
  "_gzread", referenced from:
      _gz_uncompress in minigzip.o
  "_gzwrite", referenced from:
      _gz_compress in minigzip.o
ld: symbol(s) not found for architecture x86_64
clang-14: error: linker command failed with exit code 1 (use -v to see invocation)

I have fixed it temporarily by adding -lz into the build.ninja file to the example.o's compile flags but this should be done in cmake somehow

thowell, 2023-12-22:
@keszegrobert is this issue resolved now?

thowell, 2024-01-14:
@keszegrobert this issue should be resolved. Additionally, please see my response #219.

closed 2024-01-14 by thowell

keszegrobert, 2024-01-17:
Sorry for my late respones, it seems to be fixed now