google-deepmind/mujoco_mpc #219: Linking error on MacOS 11(x86_64), not able to link mujoco with --gc-sections

keszegrobert, 2023-11-25:
This was my setup:
```
brew install llvm@14
#it took a day or so

git clone https://github.com/google-deepmind/mujoco_mpc.git
cd mujoco_mpc
mkdir build
cd build
export CC=clang
export CXX=clang++
cmake .. -G Ninja -DMJPC_BUILD_GRPC_SERVICE:BOOL=ON
ninja
```
As I have seen at: https://github.com/google-deepmind/mujoco_mpc/blob/main/.github/workflows/build.yml

```
[108/2754] Linking CXX shared library lib/libmujoco.3.0.1.dylib
FAILED: lib/libmujoco.3.0.1.dylib 
: && /usr/local/opt/llvm@14/bin/clang++ -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk -mmacosx-version-min=11.7 -dynamiclib -Wl,-headerpad_max_install_names -mmacosx-version-min=11 -Wl,-no_weak_imports -fuse-ld=lld -Wl,--gc-sections -current_version 3.0.1 -o lib/libmujoco.3.0.1.dylib -install_name @rpath/libmujoco.3.0.1.dylib _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_callback.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_box.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_convex.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_driver.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_primitive.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_collision_sdf.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_core_constraint.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_core_smooth.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_crossplatform.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_derivative.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_derivative_fd.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_forward.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_inverse.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_island.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_io.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_passive.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_plugin.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_print.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_ray.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_resource.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_sensor.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_setconst.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_solver.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_support.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_blas.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_errmem.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_misc.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_solve.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_sparse.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_util_spatial.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_vfs.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_vis_init.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_vis_interact.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_vis_state.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/engine/engine_vis_visualize.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_composite.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_flexcomp.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_mesh.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_model.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_objects.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/user/user_util.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_api.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_base.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_native_reader.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_numeric_format.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_native_writer.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_urdf.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/xml/xml_util.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/glad/glad.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/glad/loader.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/render_context.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/render_gl2.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/render_gl3.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/render/render_util.c.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/thread/thread_pool.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/thread/thread_task.cc.o _deps/mujoco-build/CMakeFiles/mujoco.dir/src/ui/ui_main.c.o  lib/libccd.a  lib/liblodepng.a  lib/libqhullstatic_r.a  lib/libtinyobjloader.a  lib/libtinyxml2.a  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk/usr/lib/libm.tbd  -lm && :
ld64.lld: error: unknown argument '--gc-sections'
clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
[113/2754] Building CXX object _deps/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
ninja: build stopped: subcommand failed.
```

I had to edit cmake/MujocoLinkOptions.cmake to not build with --gc-sections, but I think it can be done better

thowell, 2024-01-14:
The errors you were experiencing are likely related to dependency issues that have since been resolved. If you are still experiencing the issues described above please reopen this issue.

You can also remove gRPC if you don't plan to use the Python API (macOS):
```
git clone https://github.com/google-deepmind/mujoco_mpc.git
cd mujoco_mpc
mkdir build
cd build
cmake .. -G Ninja -DMJPC_BUILD_GRPC_SERVICE:BOOL=OFF
cmake --build . --config=Release --target mjpc
./bin/mjpc
```

closed 2024-01-14 by thowell

keszegrobert, 2024-01-17:
still getting errors by running the above sequence of commands
```
ld64.lld: error: unknown argument '--gc-sections'
```
This is weird because in the config step this is written out:
```
-- Performing Test SUPPORTS_GC_SECTIONS
-- Performing Test SUPPORTS_GC_SECTIONS - Success
```

keszegrobert, 2024-01-17:
I don't have permission to reopen this issue

commit e6370619 (2024-01-20): Merge pull request #253 from keszegrobert/gc-sections-linking-fix

#219 Update MujocoLinkOptions.cmake