google-deepmind/mujoco_mpc #220: The ```ctest --output-on-failure .``` command is not working as it is suggested in the CONTRIBUTING.md

keszegrobert, 2023-11-25:
My setup was:
```
user@Macbook-MacBook-Air-4 mujoco_mpc % clang --version
Homebrew clang version 14.0.6
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /usr/local/opt/llvm@14/bin
user@Macbook-MacBook-Air-4 mujoco_mpc % ctest --version
ctest version 3.27.8
```
After I successfully compiled the code by ninja, tried to run the tests, but the output was this:
```
user@Macbook-MacBook-Air-4 mujoco_mpc % cd build
user@Macbook-MacBook-Air-4 build % ctest --output-on-failure .
Test project /Users/user/Projects/mujoco_mpc/build
No tests were found!!!
```
the tests are there
```
user@Macbook-MacBook-Air-4 build % ls ./bin 
agent_server		cubic_test		exhaustive1_test	grpc_node_plugin	mjpc			regexp_test		task_test
agent_test		dfa_test		exhaustive2_test	grpc_objective_c_plugin	norm_test		required_prefix_test	threadpool_test
agent_utilities_test	direct_force_test	exhaustive3_test	grpc_php_plugin		parse_test		robust_planner_test	trajectory_test
backward_pass_test	direct_optimize_test	exhaustive_test		grpc_python_plugin	possible_match_test	rollout_test		ui_agent_server
batch_filter_test	direct_parameter_test	filter_server		grpc_ruby_plugin	protoc			sampling_planner_test	unscented_test
batch_prior_test	direct_sensor_test	filtered_re2_test	ilqg_test		protoc-3.21.12.0	search_test		utilities_test
bssl			direct_server		gradient_planner_test	kalman_test		random_test		set_test		zero_test
charclass_test		direct_trajectory_test	gradient_test		linear_test		re2_arg_test		simplify_test
compile_test		direct_utilities_test	grpc_cpp_plugin		mimics_pcre_test	re2_test		state_test
cost_derivatives_test	example			grpc_csharp_plugin	minigzip		regexp_benchmark	string_generator_test
```
 but cmake is unable to run them:
```
user@Macbook-MacBook-Air-4 build % ./bin/zero_test 
Running main() from /Users/user/Projects/mujoco_mpc/build/_deps/googletest-src/googletest/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from GradientTest
[ RUN      ] GradientTest.ZeroTest
[       OK ] GradientTest.ZeroTest (0 ms)
[----------] 1 test from GradientTest (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 1 test.
```

thowell, 2023-12-22:
Try: 
```
cd build/mjpc/test
ctest --output-on-failure .
```

This is fixed in #233.

closed 2023-12-22 by thowell

keszegrobert, 2024-01-17:
gr8, thx