google-deepmind/mujoco_mpc #227: Question regarding simultaneous agent server access

Iz-Idk, 2023-12-05:
Hi, regarding this issue #217. We have been meaning to understand it more, however, we have run into a problem that involves the mocap_pos. From what I have understood, when the agent server is called it tries to find a free port to access it (which in both ui_agent_servers does connect to different ports at the same time). Eventhough I want to give different goals to each robot, each take the last goal that it was connected to or was recently updated to. Is it because the client from the grpc is using the same server? or is there anything you could recommend us?
Here is an example code

<details>
<summary> Working example </summary>

```
import time as time_

import mujoco
import mujoco.viewer as viewer
from mujoco_mpc import agent as agent_lib
import numpy as np

import pathlib

# Creating model and data
m = mujoco.MjModel.from_xml_path("../../../../../mujoco_mpc/mjpc/tasks/quadruped/task_flat2.xml")
d = mujoco.MjData(m)

# Initializing our agent (agent server/executable)
agent = agent_lib.Agent(
        # This is to enable the ui
        server_binary_path=pathlib.Path(agent_lib.__file__).parent
        / "mjpc"
        / "ui_agent_server",
        task_id="Quadruped Flat", 
        model=m)

agent2 = agent_lib.Agent(
        # Enable the ui and finds a freeport
        server_binary_path=pathlib.Path(agent_lib.__file__).parent
        / "mjpc"
        / "ui_agent_server",
        task_id="Quadruped Flat", 
        model=m)

## Data for agents

# weights
agent.set_cost_weights({"Position": 0.15})
print("Cost weights:", agent.get_cost_weights())

# parameters
agent.set_task_parameter("Walk speed", 1.0)
print("Parameters:", agent.get_task_parameters())

# weights for second agent
agent2.set_cost_weights({"Position": 0.15})
print("Cost weights:", agent.get_cost_weights())

# parameters for second agent
agent2.set_task_parameter("Walk speed", 1.0)
print("Parameters:", agent.get_task_parameters())

# run planner for num_steps
num_steps = 8
for _ in range(num_steps):
    agent.planner_step()
    agent2.planner_step()

# Multiple goals
goals = [
    [7, 0, 0.26],
    [0, -2, 0.26],
    [-2, 0, 0.26],
    [0, -5, 0.26]
]
i = 0
goal_agent = []
goal_agent2 = []

# Settings different goals for the different robots
goal_agent = d.mocap_pos
goal_agent[0] = goals[0]
print(goal_agent)

goal_agent2 = d.mocap_pos
goal_agent2[0] = goals[1]
print(goal_agent)

with mujoco.viewer.launch_passive(m, d) as viewer:
    # Close the viewer automatically after 30 wall-seconds.

    while viewer.is_running():
        
        # set planner state
        agent.set_state(
            time=d.time,
            qpos=d.qpos[:19],
            qvel=d.qvel[:18],
            act=d.act,
            mocap_pos=goal_agent,
            mocap_quat=d.mocap_quat,
        )

        # set planner state for agent 2
        agent2.set_state(
            time=d.time,
            qpos=d.qpos[19:],
            qvel=d.qvel[18:],
            act=d.act,
            mocap_pos=goal_agent2,
            mocap_quat=d.mocap_quat,
        )

        print(d.mocap_pos) # Seems to take only in consideration the second agents goal
        # Obtain their joints from each agent and display them on the robots
        agent_ctrl = agent.get_action()
        agent_ctrl2 = agent2.get_action()
        agent_ctrl = np.append(agent_ctrl, agent_ctrl2, axis=0)
        d.ctrl = agent_ctrl
        
        # To detect when the robot is close to the goal
        if (np.linalg.norm(d.mocap_pos[0] - d.body('trunk').xpos) < 1) or (np.linalg.norm(d.mocap_pos[0] - d.body('trunk2').xpos) < 1) :
            goal_agent2[0] = goals[i] # Only updates 1 agent's goal
            print("\nARRIVED!")
            i = (i + 1) % len(goals)

        mujoco.mj_step(m,d)

        # Example modification of a viewer option: toggle contact points every two seconds.
        with viewer.lock():
            viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(d.time % 2)

        # Pick up changes to the physics state, apply perturbations, update options from GUI.
        viewer.sync()

```

</details>

A few notes are that the task_flat2 contains 2 robots with different spawning points and they both go to a same goal.  And we have also tried putting both goals to no avail. Here is a video example [Video](https://github.com/google-deepmind/mujoco_mpc/assets/74568854/2ce90f61-40d4-46e4-adef-c9eefbc540c8)
Thank you for your time and sorry for the trouble.

thowell, 2024-01-14:
Cool project!

I took a look at the code you provided and didn't immediately see any problems. Can you provide a link to: `task_flat2.xml`? This way I can try and replicate the issue on my end.

Iz-Idk, 2024-01-15:
Hi thank you for your answer!
The code for the task_flat2.xml is 
<details>
<summary> task_flat2.xml </summary>

```
<mujoco model="Quadruped">
  <include file="../common.xml"/>

  <size memory="1M"/>

  <custom>
    <!-- agent -->
    <text name="task_transition" data="Quadruped|Biped|Walk|Scramble|Flip" />
    <numeric name="agent_planner" data="2 " />
    <numeric name="agent_horizon" data="0.35" />
    <numeric name="agent_timestep" data="0.01" />
    <numeric name="sampling_spline_points" data="3" />
    <numeric name="sampling_trajectories" data="60"/>
    <numeric name="sampling_exploration" data="0.04" />
    <numeric name="gradient_spline_points" data="5" />

    <!-- residual -->
    <numeric name="residual_select_Gait" data="0"/>
    <text name="residual_list_Gait" data="Stand|Walk|Trot|Canter|Gallop"/>
    <numeric name="residual_select_Gait switch" data="1"/>
    <text name="residual_list_Gait switch" data="Manual|Automatic"/>
    <numeric name="residual_Cadence" data="2 0 4" />
    <numeric name="residual_Amplitude" data=".06 0 0.2" />
    <numeric name="residual_Duty ratio" data="0 0 1"/>
    <numeric name="residual_Walk speed" data="0 0 4"/>
    <numeric name="residual_Walk turn" data="0 -2 2"/>
    <numeric name="residual_select_Flip dir" data="0"/>
    <text name="residual_list_Flip dir" data="Back Flip|Front Flip"/>
    <numeric name="residual_select_Biped type" data="0"/>
    <text name="residual_list_Biped type" data="Foot Stand|Hand Stand"/>
    <numeric name="residual_Heading" data="0 -3.14 3.14" />

    <!-- estimator -->
    <numeric name="estimator" data="0" />
    <numeric name="estimator_sensor_start" data="9" />
    <numeric name="estimator_number_sensor" data="21" />
    <numeric name="estimator_timestep" data="0.005" />
    <numeric name="batch_configuration_length" data="3" />
    <numeric name="batch_scale_prior" data="1.0e-3" />
  </custom>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <global azimuth="-90"/>
  </visual>

  <statistic extent="1" center="0 0 .3"/>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" diffuse="0.5 0.5 0.5" specular="0.3 0.3 0.3"
      directional="true" castshadow="false"/>
    <geom name="floor" size="0 0 0.1" pos="0 0 -0.01" type="plane" material="blue_grid"/>

     <body name="goal" mocap="true" pos=".4 0.4 0.3">
        <geom size="0.12" contype="0" conaffinity="0" rgba="0 1 0 .5" group="2" />
      </body>
      
      <body name="box" mocap="true" pos="-2.5 0 0">
        <geom name="box" class="prop" size="1 1 0.3"/>
      </body>
      <geom name="ramp" class="prop" pos="3.13 2.5 -.18" size="1.6 1 .5" euler="0 -0.2 0"/>
      <geom name="hill" class="prop" pos="6 6 -5.5" size="6" type="sphere"/>
  </worldbody>

[... 487 lines cut ...]
          <joint class="hip" name="RL_thigh_joint2"/>
          <geom class="visual" mesh="thigh"/>
          <geom class="thigh1"/>
          <geom class="thigh2"/>
          <geom class="thigh3"/>
          <body name="RL_calf2" pos="0 0 -0.2">
            <inertial mass="0.226" pos="0.00472659 0 -0.131975"
                quat="0.706886 0.017653 0.017653 0.706886"
                diaginertia="0.00340344 0.00339393 3.54834e-05"/>
            <joint class="knee" name="RL_calf_joint2"/>
            <geom class="visual" mesh="calf"/>
            <geom class="calf1"/>
            <geom class="calf2"/>
            <geom name="HL2" class="foot"/>
            <site name="RL2" pos="0 0 -0.2" type="sphere" size=".015"/>
          </body>
        </body>
      </body>
    </body>
  </worldbody>
</mujoco>
```
</details>

Thank you again for your help :)

thowell, 2024-01-15:
A couple things to look into:

- the keyframe is incorrect. the nominal quaternion should be `1 0 0 0` for each quadruped.
- the main script doesn't call `planner_step` in the main loop, this routine should to be called at least once for each quadruped before a simulation step.

If you think the mocap goal position is the issue, I recommend calling `get_state` after `set_state` to ensure that the mocap input to the agent was valid.