google-deepmind/mujoco_mpc #239: build error in  brace-enclosed initializer list for mjuDef.

Rian-Jo, 2024-01-10:
Hi, happy new year.

First of all, I'm the fan of your works. Thank you.

In my computer, i cloned the repository and compiled. Then, when i build, i got the error message for 

[build] /mujoco_mpc/mjpc/agent.cc:555:14: error: no match for ‘operator=’ (operand types are ‘mjuiDef’ {aka ‘mjuiDef_’} and ‘<brace-enclosed initializer list>’)
[build]   555 |         "0 1"};

and other codes using mjuiDef types.

So i changed from

'''
defFeatureParameters[i + parameter_shift] = {
    mjITEM_SLIDERNUM, "residual", 2, DataAt(ActiveTask()->parameters, i),
    "0 1"
};
'''

to

'''
defFeatureParameters[i + parameter_shift] = mjuiDef{
    mjITEM_SLIDERNUM, "residual", 2, DataAt(ActiveTask()->parameters, i),
    "0 1"
};
'''

for all cases and worked properly.

Is it correct fix?

thowell, 2024-01-10:
What operating system and compiler are you using? I don't see this issue on macOS using clang 14.

Rian-Jo, 2024-01-11:
i am using gcc 9.4.0 on Ubuntu 20.04

thowell, 2024-01-11:
Try building with clang. You can find additional specifics [here](https://github.com/google-deepmind/mujoco_mpc/blob/6a56e33e37d66383a231cd5b48b26362628c3fb2/.github/workflows/build.yml#L15).

Rian-Jo, 2024-01-11:
I was inteded to use gcc..

ok i'll check. thanks for help!!