google-deepmind/mujoco_mpc #271: Iteration timing graph, total time confusion?

DMackRus, 2024-01-31:
Hi,

On the timings graph in the bottom right, is the total time line just simply the total iteration time?

When I time the "Iteration" function, the total time for that function is a good 30 % quicker than the graph seems to imply? Is there some other timing segment I am missing that is included in total time?

erez-tom, 2024-01-31:
I'm not sure what you mean.  The data presented in the lower right plot under the label "Total" [comes from](https://github.com/google-deepmind/mujoco_mpc/blob/main/mjpc/agent.cc#L1074) `agent_compute_time_`, which is computed [here](https://github.com/google-deepmind/mujoco_mpc/blob/main/mjpc/agent.cc#L278) by measuring the wall clock time of a single planning iteration.

Are you using some other timing mechanism?

DMackRus, 2024-01-31:
Hello,

Oh i see, thank you!

I was just running a timer in the "Iteration" function in the planner, I didn't realise that the nominal trajectory was computed before this function. The nominal computation time takes account of the time difference.

closed 2024-01-31 by DMackRus