google-deepmind/mujoco_mpc #277: Question about A, B C and D matrices ordering in memory?

DMackRus, 2024-02-04:
Hello,

I am trying to fix some bugs in my code and want to make sure one of my assumptions is correct.

I **think** that the A, B, C and D matrices are ordered in memory column-wise rather than row-wise? I.e lets say you have a 4x4 A matrix
The elements would be ordered like

0  4   9    13
1  5   10  14
2  6   11  15
3  7   12  16

Is this correct?

thowell, 2024-02-04:
The A, B, C, D matrices, computed [here](https://github.com/google-deepmind/mujoco/blob/66a8bef6fa7321f75878dc0f73102265309e6873/src/engine/engine_derivative_fd.c#L305) as (DyDq, DyDv, DyDa), DyDu, (Dsdq, DsDv, DsDa), DsDu are ordered column-wise (or transposed).

yuvaltassa, 2024-02-05:
As a side note, there is a currently unexposed function "mjd_stepFD" which does use column-major, but then everything is [transposed](https://github.com/google-deepmind/mujoco/blob/ced37ffbb237584512311b041bce3124e3b2cc2a/src/engine/engine_derivative_fd.c#L583), in case you want column-major for some reason (though transpose is very cheap.)

closed 2024-02-05 by yuvaltassa