google-deepmind/mujoco_mpc #280: MJPC on Ubuntu

vikashplus, 2024-02-07:
I'm facing compilation issues with MJPC on Ubuntu. I followed the build instructions mentioned [here](https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#installation). Here are all the options that I tried 
- Branch: [main](https://github.com/google-deepmind/mujoco_mpc) / [v0.0.1 release tag](https://github.com/google-deepmind/mujoco_mpc/releases/tag/v0.0.1)
- Compiler: GCC / Clang

I'm curious if the Ubuntu build is supported at the moment. 
The error logs are build related, quite long, and different for different settings. Please let me know if it will help to add the logs here.

vikashplus, 2024-02-07:
Update: So the secret sauce so far is dba4bd8d51b2233e637e00e1625593890bf4abaf and Clang 12.0.0

yuvaltassa, 2024-02-07:
Would you like to send a PR to update the documentation?

vikashplus, 2024-02-07:
It's a workaround for now. @thowell  is getting his Ubuntu machine soon. Expect a PR  once we have a watertight solution that will work more broadly.

thowell, 2024-02-10:
I just did a fresh install of Ubuntu 23.10 and didn't run into any build problems with MJPC when following the instructions on the README. I used clang 16.0.6.

vikashplus, 2024-02-10:
I am on Ubuntu 20.04

A few thoughts
- Seems like there is a dependency on a compiler. 
- Should we clarify which compiler to use?
- Should we add compiler install instructions to read me? 
> sudo apt-get install clang-12

thowell, 2024-02-12:
Tested with a fresh install of Ubuntu 22.04 with clang-12 and that worked as well. I will update the README to include this step and to look at the build.yml if troubleshooting an install.

thowell, 2024-02-12:
@vikashplus should we add any additional instructions?  #286

vikashplus, 2024-02-12:
Clang 12 is the primary thing for the setup I tested for. Let's merge for now. I'll update you if I learn anything new.

richardrl, 2024-03-09:
I think we should put a note on the main page to use Clang12, Clang12++.

I did the following to get things to compile on ubuntu 20. with gcc, it errors around 70%

```
mkdir build
cd build
sudo apt-get install clang-12
CXX=clang12++ CC=clang12 cmake ..
make -j 5
```

thowell, 2024-05-19:
@vikashplus please see #324