google-deepmind/mujoco_mpc #293: Compilation issue with gcc (11.4)

DwarKapex, 2024-02-14:
Hi all,

  I see this issue when try to compile MPC with GNU compiler 11.4:
  ```
#10 912.6         /opt/mujoco-mpc/mjpc/../mjpc/spline/spline.h:106:9: error: expected unqualified-id before ‘const’
#10 912.6           106 |         const IteratorT<SplineType, NodeType>& other) = default;
#10 912.6               |         ^~~~~
#10 912.6         /opt/mujoco-mpc/mjpc/../mjpc/spline/spline.h:105:37: error: expected ‘)’ before ‘const’
#10 912.6           105 |     IteratorT<SplineType, NodeType>(
#10 912.6               |                                    ~^
#10 912.6               |                                     )
#10 912.6           106 |         const IteratorT<SplineType, NodeType>& other) = default;
#10 912.6               |         ~~~~~
#10 912.6         /opt/mujoco-mpc/mjpc/../mjpc/spline/spline.h:109:68: error: expected ‘)’ before ‘&&’ token
#10 912.6           109 |     IteratorT<SplineType, NodeType>(IteratorT<SplineType, NodeType>&& other) =
#10 912.6               |                                    ~                               ^~
#10 912.6               |                                                                    )
```

But all is fine when I force cmake to use clang.

thowell, 2024-02-15:
Thanks for opening this issue. This should be fixed soon. Currently, `spline.h` and `spline.cc` are not used in MJPC so you can comment out these files in the CMakeList if you need to build with gcc.

thowell, 2024-05-19:
This issue should be resolved. Installation instructions (#324) now specify to build with clang.

closed 2024-05-19 by thowell