google-deepmind/mujoco_mpc #300: Clarification on Planner Running Frequency and Behavior in MuJoCo MPC Code

jc-bao, 2024-03-01:
First off, kudos to the MuJoCo team for the excellent MPC code! I'm delving into the details and have a question regarding the planner's operational frequency within the code. It appears there's no explicit sleep function or scheduling mechanism present, leading me to wonder if the planner operates at maximum possible speed, rather than adhering to a predetermined running frequency. Specifically, I'm curious if, between two consecutive knot points in the planned trajectory, the planner might execute multiple iterations depending on the hardware capabilities.

Furthermore, concerning the approach of varying frequency replanning, does each iteration involve reinitializing the nominal trajectory (mainly the new knot points) by resampling from the previous plan?

Any insights or information on this matter would be greatly appreciated!

thowell, 2024-03-01:
When using the MJPC GUI, planning is performed asynchronously. This means the planner will replan (without waiting) as fast as your specific hardware allows.

The planner is warmstarted with the solution from the previous iteration. The previously solution is either resampled or rolled out given the current state, depending on the planner.

jc-bao, 2024-03-01:
Got it. Really smart design! Thank you so much for the clarification!

closed 2024-03-01 by jc-bao