google-deepmind/mujoco_mpc #310: MJPC for parallel manipulators

ahip88, 2024-03-22:
I recall that Mujoco wasn't made with the intent for parallel manipulators. Is this still true?
Stewart platforms with MJPC take a very long time for converging to a certain pose/acceleration. Model-free RL works fine however (with Mujoco) so I'm unsure if it's something with my setup (cost etc.) or something more inherent to the simulator when using MPC.
Advice appreciated. Thank you!

ahip88, 2024-03-22:
I forgot here's the model I'm using:
self._plat = """
    <mujoco model="plat">
    
     <default>
    <joint armature="10e-3" limited="false" />
    <geom margin="10e-3" rgba="0.97 0.38 0.06 1" />
  </default>

  <compiler inertiafromgeom="true" angle="degree" fusestatic="true" autolimits="true"/>
  <size njmax="800" nconmax="400"/>
  <option integrator="RK4" timestep="0.0125">
    <flag sensornoise="enable" multiccd="enable" contact="enable" energy="enable"/>
  </option>

  <visual>
      <map zfar="30"/>
      <rgba haze="0.15 0.25 0.35 1"/>
      <quality shadowsize="128"/>
      <global offwidth="100" offheight="100"/>
  </visual>

  <asset>
      <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/> 
      <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
      <texture name="texgeom" type="cube" builtin="flat" mark="cross" width="127" height="1278" 
          rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" markrgb="1 1 1" random="0.01"/>  

      <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
      <material name="matgeom" texture="texgeom" texuniform="true" rgba="0.8 0.6 .4 1"/>
      <mesh file="seat_4_scaled.stl" />
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -0.15" size="0 0 .25" type="plane" material="matplane" contype='0' condim='3'/>

    <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>
    <light mode="targetbodycom" target="mobile_plat" directional="false" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 0 4.0" dir="0 0 -1"/>

    <camera name="closeup" mode="targetbodycom" target="mobile_plat" pos="0 -1.8 0.5"/>
	<body name="humanoid_mass" pos="0 0.01 0.175">
		<site name="invis_1" pos="0 0.01 0.09" rgba="0 0 0 0" />
		<joint axis="0 0 1" name="humanoid_mass_z" pos="0.0 0.0 0.0" range="-100 100" type="slide"/>
		<joint axis="0 1 0" name="humanoid_mass_y" pos="0.0 0.0 0.0" range="-100 100" type="slide"/>
		<joint axis="1 0 0" name="humanoid_mass_x" pos="0.0 0.0 0.0" range="-100 100" type="slide"/>
		<joint axis="0 0 1" name="humanoid_mass_rot_z" pos="0.0 0.0 0.0" range="-180 180" type="hinge"/>
		<joint axis="0 1 0" name="humanoid_mass_rot_y" pos="0.0 0.0 0.0" range="-180 180" type="hinge"/>
		<joint axis="1 0 0" name="humanoid_mass_rot_x" pos="0.0 0.0 0.0" range="-180 180" type="hinge"/>
		<geom name="humanoid_mass_geom" pos="0 0.02 0.0445" size="0.01 0.01 0.01" mass="17.0" type="box" rgba="1 1 1 1"/>
	</body>	
	

	<body name="sensor_orientation_body" euler="0 0 0"  pos="0 0 0.1" >
        <joint axis="0 0 1" name="sensor_orientation_z" type="slide"/>
        <joint axis="0 1 0" name="sensor_orientation_y" type="slide"/>
        <joint axis="1 0 0" name="sensor_orientation_x" type="slide"/>
        <joint axis="0 0 1" name="sensor_orientation_rot_z" type="hinge"/>
        <joint axis="0 1 0" name="sensor_orientation_rot_y" type="hinge"/>
        <joint axis="1 0 0" name="sensor_orientation_rot_x" type="hinge"/>
        <site name="invis_3" pos="0.0 0.0 0.0" type="sphere" size="0.01" rgba="0 0 0 0"/>
        <geom name="sensor_geom" pos="0 0 0" size="0.01 0.01 0.01" type="box" rgba="1 1 1 1"/>
    </body>
	
    <body name="mobile_plat" pos="0 0 0.1933" >
	  <site name="invis_2" pos="0 0.05 0.1" rgba="0 0 0 0"/>
	  <site name="invis_2_2" pos="0 -0.01 0.04" rgba="0 0 0 0" />
      <site name="root"/>
	  <geom name="seat" type="mesh" mesh="seat_4_scaled" mass="10" pos="0 0 0.04" gap ="0.5"/>
      <geom name="mobile_plat_geom" pos="0 0 0" size="0.29 0.0075" mass="15.00" type="cylinder" rgba="0.1 0.1 0.1 1"/>
	  <joint axis="0 0 1" name="plat_z" pos="0.0 0.0 0.0" range="-100 100" type="slide"/>
[... 142 lines cut ...]
		<weld body1="sensor_orientation_body" body2="mobile_plat"/>
        <weld body1="motor1" body2="motor2"></weld>
        <weld body1="motor2" body2="motor3"></weld>
        <weld body1="motor3" body2="motor4"></weld>
        <weld body1="motor4" body2="motor5"></weld>
        <weld body1="motor5" body2="motor6"></weld>
        <weld body1="motor6" body2="motor1"></weld>
        <weld body1="motor1" body2="static"></weld>
		<weld body1="motor2" body2="static"></weld>
		<weld body1="motor3" body2="static"></weld>
		<weld body1="motor4" body2="static"></weld>
		<weld body1="motor5" body2="static"></weld>
		<weld body1="motor6" body2="static"></weld>
		
  </equality>  
  <sensor>
    <accelerometer name="mobile_plat_accel" noise="0.03924" site="invis_3"/>
    <gyro name="mobile_plat_gyro" noise="0.00003" site="invis_3"/>
    <magnetometer name="mobile_plat_mag" noise="0.25" site="invis_3"/>
  </sensor>
</mujoco>

    """
    
   The goal is to manipulate the sensor via. motor positions.

ahip88, 2024-03-22:
...
<body name="static" pos="0 0 0">
      <geom name="static_geom" pos="0 0 -0.125" size="0.696 0.02" mass="10e7"  type="cylinder" rgba="0.04 0.04 0.04 1"/>
    </body>
  </worldbody>
  <tendon>
    <spatial stiffness="1000" width="0.001" damping="100" limited="false" >
        <site site="invis_1"/>
        <site site="invis_2"/>
    </spatial>
	<spatial stiffness="1000" width="0.001" damping="100" limited="false" >
        <site site="invis_1"/>
        <site site="invis_2_2"/>
    </spatial>
</tendon>
  <actuator>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_1_2"/>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_2_2"/>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_3_2"/>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_4_2"/>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_5_2"/>
    <position ctrlrange="-1.0 1.0" kp="25" joint="ankle_6_2"/>
  </actuator>
  <equality>
		<weld body1="sensor_orientation_body" body2="mobile_plat"/>
        <weld body1="motor1" body2="motor2"></weld>
        <weld body1="motor2" body2="motor3"></weld>
        <weld body1="motor3" body2="motor4"></weld>
        <weld body1="motor4" body2="motor5"></weld>
        <weld body1="motor5" body2="motor6"></weld>
        <weld body1="motor6" body2="motor1"></weld>
        <weld body1="motor1" body2="static"></weld>
		<weld body1="motor2" body2="static"></weld>
		<weld body1="motor3" body2="static"></weld>
		<weld body1="motor4" body2="static"></weld>
		<weld body1="motor5" body2="static"></weld>
		<weld body1="motor6" body2="static"></weld>
		
  </equality>  
  <sensor>
    <accelerometer name="mobile_plat_accel" noise="0.03924" site="invis_3"/>
    <gyro name="mobile_plat_gyro" noise="0.00003" site="invis_3"/>
    <magnetometer name="mobile_plat_mag" noise="0.25" site="invis_3"/>
  </sensor>
</mujoco>

    """

thowell, 2024-03-26:
Can you please link to your fork of MJPC and the task you have defined?

closed 2024-04-10 by ahip88