google-deepmind/mujoco_mpc #311: Reproducing Rubik's cube

richardrl, 2024-03-27:
Hi all,
I am unable to reproduce performance on Rubik's cube with default settings. I also played around with it and typically very little movement happens, a couple fingers just wiggle a bit but cube doesn't really move.

I have heard this is possible because my computer is not fast enough.

If so, is there some setting to freeze real time simulation, but compute the optimal plan offline with a certain number of compute, then play it back in the sim?

thowell, 2024-03-27:
The video results for this task are generated with a MacBook Pro with M1 Pro chip.

Make sure you build MJPC in Release mode. For my laptop I needed to change the real time rate to less than 50% (you can do this with '-' key). You can also set the number of rollout trajectories to 10 if your machine has less CPU cores.

richardrl, 2024-03-27:
Thank you for the reply I will try the minus sign to slow things down.
I was unable to get the vscode instructions working so instead I did:

mkdir build
cmake ..
make .

Is this release mode or do we need more settings?

richardrl, 2024-04-04:
Yes I saw those instructions but was not able to follow it. 
Can you provide more instructions?

1. I think the first time I opened the project in VSCode I saw the configuration popup, but I don't see it anymore and not sure how to make it come back.
2. I clicked the wrench to change the activekit to clang12 (this is a must in Ubuntu 20 or it doesnt build). Then I change the button to the right of build to release, and click build. But I get this:
[image]

Even though I set the build target as release, when I check the string, it says this:
`[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang-12 -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++-12 -S/samsung4tb/mujoco_mpc -B/samsung4tb/mujoco_mpc/build -G Ninja
`

and at the end it still errors.

I can use this command manually in terminal to build it, but not sure if I am missing performance optimizations necessary to reproduce:
`rli14@fitzroy:/samsung4tb/mujoco_mpc/build$ CXX=/usr/bin/clang++-12 CC=/usr/bin/clang-12 cmake -DCMAKE_BUILD_TYPE=Release ..
`

thowell, 2024-04-06:
Send me an email so we can get on a call this week and I can help you figure out the build.

closed 2024-05-19 by thowell