google-deepmind/mujoco_mpc #313: Questions regarding using stateful actuator intvelocity

MiaoDragon, 2024-04-03:
Hi,

I'm a PhD student trying to use this package as a library for controlling task in mujoco. 
First thanks a lot for this package. I have successfully integrated it and be able to solve example tasks in mujoco. However, I have some questions regarding using this with stateful actuator, such as `intvelocity`.
What I'm trying to do is to use `intvelocity` for tracking the position and velocity input. However, I found that mujoco-mpc does not treat `data->act` as part of the control, but rather as a state. Hence, I'm not able to sample the position control in `data->act` for my task.
I'd like to ask your input for my issue. And could you recommend any way to control the position and velocity at the same time for MPC task? I found that `intvelocity` performs better than a combination of `position` and `velocity` actuators, since I assume it's the correct way of implementing PID controller.

Thanks. I would really appreciate any help!

(I also opened another issue under mujoco at [issue](https://github.com/google-deepmind/mujoco/issues/1567))

thowell, 2024-04-04:
Since Yuval has responded on the MuJoCo thread I will close this issue.

closed 2024-04-04 by thowell

ppap36, 2024-10-28:
Hello,sorry to bother, I am controlling the robot arm by velocity actuator, but it seems that it did not go well.
Can I ask you some questions?