google-deepmind/mujoco_mpc #317: Building python API stuck at `Using FetchContent to retrieve gRPC` step

zhuoqun-chen, 2024-05-06:
# Description

C/C++ target `mjpc` already built and I'm trying to build python API via:

```
cd python
python setup.py install
```

however, the terminal got stuck at the following step forever:

```
-- mujoco::FindOrFetch: Using FetchContent to retrieve `gRPC`
```

# Debug info

- Mac M1 Pro with `Sonoma 14.4`
- Mujoco version `3.1.4`
- Python version `3.12.2`

# Full Log
```
/Users/zqchen/micromamba/envs/mujoco_mpc/lib/python3.12/site-packages/setuptools/__init__.py:81: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
!!

        ********************************************************************************
        Requirements should be satisfied by a PEP 517 installer.
        If you are using pip, you can try `pip install --use-pep517`.
        ********************************************************************************

!!
  dist.fetch_build_eggs(dist.setup_requires)
running install
/Users/zqchen/micromamba/envs/mujoco_mpc/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/Users/zqchen/micromamba/envs/mujoco_mpc/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
creating mujoco_mpc.egg-info
writing mujoco_mpc.egg-info/PKG-INFO
writing dependency_links to mujoco_mpc.egg-info/dependency_links.txt
writing requirements to mujoco_mpc.egg-info/requires.txt
writing top-level names to mujoco_mpc.egg-info/top_level.txt
writing manifest file 'mujoco_mpc.egg-info/SOURCES.txt'
reading manifest file 'mujoco_mpc.egg-info/SOURCES.txt'
writing manifest file 'mujoco_mpc.egg-info/SOURCES.txt'
[... 116 lines cut ...]

-- mujoco::FindOrFetch: Using FetchContent to retrieve `ccd` - Done
-- mujoco::FindOrFetch: checking for targets in package `mujoco`
-- mujoco::FindOrFetch: checking for targets in package `mujoco` - found
-- mujoco::FindOrFetch: checking for targets in package `glfw3`
-- mujoco::FindOrFetch: checking for targets in package `glfw3` - target `glfw` not defined.
-- mujoco::FindOrFetch: Using FetchContent to retrieve `glfw3`
-- Using Cocoa for window creation
-- mujoco::FindOrFetch: Using FetchContent to retrieve `glfw3` - Done
-- mujoco::FindOrFetch: Using FetchContent to retrieve `mujoco` - Done
-- mujoco::FindOrFetch: checking for targets in package `GTest`
-- mujoco::FindOrFetch: checking for targets in package `GTest` - target `gtest` not defined.
-- mujoco::FindOrFetch: Using FetchContent to retrieve `googletest`
-- mujoco::FindOrFetch: Using FetchContent to retrieve `googletest` - Done
-- mujoco::FindOrFetch: checking for targets in package `absl`
-- mujoco::FindOrFetch: checking for targets in package `absl` - target `absl::core_headers` not defined.
-- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp`
-- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp` - Done
-- mujoco::FindOrFetch: checking for targets in package `glfw`
-- mujoco::FindOrFetch: checking for targets in package `glfw` - found
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libz.tbd (found version "1.2.12")
-- mujoco::FindOrFetch: checking for targets in package `gRPC`
-- mujoco::FindOrFetch: checking for targets in package `gRPC` - target `gRPC` not defined.
-- mujoco::FindOrFetch: Using FetchContent to retrieve `gRPC`
```

yuvaltassa, 2024-05-06:
How long did you wait? gRPC is unfortunately a very large dependency...

thowell, 2024-05-19:
@zhuoqun-chen this PR #324 notes that gRPC installation can take some time ~10-20 minutes.

closed 2024-05-19 by thowell