google-deepmind/mujoco_mpc #320: Request: open Discussion forum on this repository?

MyelinsheathXD, 2024-05-08:
Hello everybody! Hey, this is amazing tech and need to be have a conversational questions about the system rather than only  issues on the repository.

It would be greate if there is a  Discussion forum on this repository like mujoco repository.
Could you enable Discussion forum on this repository?

yuvaltassa, 2024-05-10:
I tried, I swear. Couldn't figure it out 😛 
Paging @nimrod-gileadi

shivanshpatel35, 2024-10-30:
I also feel that a discussion forum will be super helpful. The overall codebase is more complex than other simulation codebases, so a discussion forum will help reach this work to more people faster.

yuvaltassa, 2024-10-31:
@nimrod-gileadi, @thowell, how do we do this?

ShahhhVihaan, 2024-11-05:
Yup, I would love this too!

I understand that the repository currently supports shooting-based planners, which are ideal for real-time applications. However, I’m wondering if you would consider adding collocation methods as an option for users who need higher trajectory accuracy, even if it means sacrificing real time performance. If so, I would love to contribute.

I’m relatively new to optimal control and would like to implement both shooting and collocation methods (direct and indirect) from scratch, inspired by MJPC’s structure. I could fork MJPC and work on it independently or, if you’re interested, I’d be happy to contribute directly to this repo.

If this is something you’d like to discuss, we could explore it further in the discussion forum.

joaocandre, 2025-10-17:
@ShahhhVihaan I believe the Direct Optimization pipeline described [in the docs](https://github.com/google-deepmind/mujoco_mpc/blob/main/docs/DIRECT.md) is a sort of collocation strategy, or at least can be extended upon to get a general baseline for collocation implementations.

But the documentation could be improved, at least with an implementation MWE.