google-deepmind/mujoco_mpc #321: python API FAILED: bin/agent_server FAILED: bin/ui_agent_server

dhockingofthegods, 2024-05-10:
home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/__init__.py:81: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
!!

        ******************************************************************************
        Requirements should be satisfied by a PEP 517 installer.
        If you are using pip, you can try pip install --use-pep517.
        ******************************************************************************

!!
  dist.fetch_build_eggs(dist.setup_requires)
running install
/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ******************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ******************************************************************************

!!
  self.initialize_options()
/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ******************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ******************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
writing mujoco_mpc.egg-info/PKG-INFO
writing dependency_links to mujoco_mpc.egg-info/dependency_links.txt
writing requirements to mujoco_mpc.egg-info/requires.txt
writing top-level names to mujoco_mpc.egg-info/top_level.txt
reading manifest file 'mujoco_mpc.egg-info/SOURCES.txt'
writing manifest file 'mujoco_mpc.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running generate_proto_grpc
touch /home/dhocking/mjc/mujoco_mpc/python/build/lib.linux-x86_64-cpython-312/mujoco_mpc/proto/__init__.py
running copy_task_assets
copying mujoco_mpc/filter_test.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/direct.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/direct_test.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/agent.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/filter.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/ui_agent_test.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/mjpc_parameters.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/__init__.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
copying mujoco_mpc/agent_test.py -> build/lib.linux-x86_64-cpython-312/mujoco_mpc
running build_ext
Configuring CMake with the following arguments:
  -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE
  -DCMAKE_BUILD_TYPE:STRING=Debug
  -DBUILD_TESTING:BOOL=OFF
  -DMJPC_BUILD_GRPC_SERVICE:BOOL=ON
CHECK_STARTmujoco::FindOrFetch: checking for targets in package mujoco
CHECK_FAILtarget mujoco::mujoco not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve mujoco
CHECK_STARTmujoco::FindOrFetch: checking for targets in package qhull
[... 214 lines cut ...]
    cmd_obj.run()
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/_distutils/command/install_lib.py", line 110, in build
    self.run_command('build_ext')
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
    self.distribution.run_command(command)
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
    super().run_command(command)
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/dhocking/mjc/mujoco_mpc/python/setup.py", line 191, in run
    self._configure_and_build_agent_server()
  File "/home/dhocking/mjc/mujoco_mpc/python/setup.py", line 232, in _configure_and_build_agent_server
    subprocess.check_call(
  File "/home/dhocking/anaconda3/envs/env1_3_11/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '/home/dhocking/mjc/mujoco_mpc/build', '--target', 'agent_server', 'ui_agent_server', '-j8', '--config', 'Debug']' returned non-zero exit status 1.
I can run the GUI but when i type
python setup.py install it have some issue like that
[5/6] Linking CXX executable bin/agent_server
FAILED: bin/agent_server 
[6/6] Linking CXX executable bin/ui_agent_server
FAILED: bin/ui_agent_server 
i have fix this bug for 4 days and have no ideal what to do to fix this

dhockingofthegods, 2024-05-11:
i have fix this bug, i upgrade ubuntu 20.04 to 22.04 and use clang-14

thowell, 2024-05-19:
Please see improved installation instructions: #324

closed 2024-06-05 by thowell