google-deepmind/mujoco_mpc #326: Control costs = 0 in demo

mikezhang95, 2024-05-29:
Hi MJPC team: 

When I am running the cartpole demo (https://github.com/google-deepmind/mujoco_mpc/blob/main/python/mujoco_mpc/demos/agent/cartpole.py), I find that the cost term of the control is always 0. Is it as designed?

thowell, 2024-05-29:
The weighted control term is small. If you increase the weight (eg, to 1.0) in the GUI you can see this term in the Objective live plot.

mikezhang95, 2024-05-29:
Thank you very much for your quick reply! 
I am using `cartpole.py` in the demos. I think the variable `cost_terms` is achieved from the function `get_cost_term_values`, which is not weighted. I think the problem comes from the `agent.planner_step` doesn't update `data.ctrl`.

thowell, 2024-05-30:
It looks like you are correct that we don't set `data->ctrl` in the gRPC interface with an action from the policy. The mjData object [here](https://github.com/google-deepmind/mujoco_mpc/blob/0e5a7b8387b777c650d9624a6bf5798b1a462ec4/mjpc/grpc/grpc_agent_util.cc#L268) needs to have the correct `ctrl` in order for the cost term to be computed correctly.

thowell, 2024-06-02:
@mikezhang95 this bug should be addressed with #327.

mikezhang95, 2024-06-03:
Thank you for your great work!

closed 2024-06-03 by thowell