google-deepmind/mujoco_mpc #332: Missing XML File

deniz0425, 2024-07-10:
Hey,
I was trying to implement the predictive sampling planner in the MJX module inside the python section. I tried running the existing example that was set to mjx_insert_cube.xml inside the insert.py file [https://github.com/google-deepmind/mujoco_mpc/blob/main/python/mujoco_mpc/mjx/tasks/insert.py](url) , but the xml file did not exist in the repository.

thowell, 2024-07-10:
Yes, it looks like we are missing this file.

@erez-tom is this file created from a Menagerie model?

deniz0425, 2024-07-18:
Hey, 
Just wondering if that mjx_insert_cube.xml file was able to be found. I looked at [https://github.com/google-deepmind/mujoco_menagerie](url) and couldn't find it there either.

oscarkfpang, 2025-03-29:
Try this:
[https://github.com/google-deepmind/mujoco_menagerie/blob/main/franka_emika_panda/mjx_single_cube.xml](https://github.com/google-deepmind/mujoco_menagerie/blob/main/franka_emika_panda/mjx_single_cube.xml)