google-deepmind/mujoco_mpc #334: Multi-agent MPC

hosseinfeiz, 2024-07-24:
Hi,

In the case of multi-humanoid tracking, is it possible to instantiate several agents, associate separate planners to each agent, and use ILQG to minimize the tracking objectives while considering future interaction contacts? I assume a higher-level agent is needed to coordinate the contacts between multiple agents.

thowell, 2024-07-27:
To consider interaction between humanoids, one option would be to create a single MuJoCo model that contains $N$ humanoids and use the existing iLQG planner to jointly optimize all humanoid trajectories. With this setup, interactions between humanoids will be considered. However, the planning problem will have complexity $O(N^3 n_u^3 T)$.

Alternatively, as you suggest, individual planners could be used for each humanoid, but then interaction between humanoids would need to be handled by a separate mechanism. One option would be to consider the motions of the other humanoids fixed (they could be modeled as time-varying but fixed geoms in the scene), and optimize trajectories for a single humanoid at a time. Iterating this scheme over all humanoids might converge to a reasonable result.

Both of these options could use MJPC as is, without a new/custom planner.

hosseinfeiz, 2024-08-05:
Thanks, it worked perfectly.

closed 2024-08-05 by hosseinfeiz