google-deepmind/mujoco_mpc #337: The problem of adding new motion capture data for human motion tracking

rottenapple777, 2024-08-08:
In the human motion tracking task, I want to add new motion capture data, but the amc file I downloaded from CMU is different from the xml of the keyframe in mjpc. Is some format conversion program being used? How do I get the new motion capture data file in the new xml format. Thanks a lot.

thowell, 2024-08-08:
@hartikainen

hartikainen, 2024-08-09:
Unfortunately, I haven't open-sourced any of these functions as they're overly complicated and not very user friendly for anyone else than myself. It's been long on my todo-list to clean up the IK-code but I'm honestly not sure if I'll ever get there.

You need to transform the `.amc`-files into our `.xml` format. For that, you basically have to extract the cartesian joint/body positions for each frame and order them like they're used here: https://github.com/google-deepmind/mujoco_mpc/blob/1f78168912b4fe4af30365198c9f4e3ecf025ba1/mjpc/tasks/humanoid/tracking/tracking.cc#L69-L73

That should be relatively easy to do. Just note that you'll have to rescale the positions with a constant value to get them right. Then just write the each frame into an `.xml` file like here: https://github.com/google-deepmind/mujoco_mpc/blob/1f78168912b4fe4af30365198c9f4e3ecf025ba1/mjpc/tasks/humanoid/tracking/keyframes/CMU-CMU-02-02_04_poses.xml#L3

That is, each `key` in the `.xml` contains `48 = 16 * 3` (cartesian xyz position for the 16 joints) `mpos` values.

What's a bit trickier is if you also want to use reset positions for the motions. These are controlled by the first `keyframe`'s `qpos` and `qvel` values (like in the above example `key`). For these, you're gonna have to do some inverse kinematics because the `.amc`-files don't include those for our humanoid. For the inverse kinematics, I've used the multi-site IK functions from here: https://github.com/google-deepmind/dm_control/pull/399.

hartikainen, 2024-08-09:
I looked around and noticed that I had opened a PR for converting the AMASS files into our `.xml` format here: https://github.com/hartikainen/mocap-environments/pull/6/files. That might give you some ideas on how to approach the conversion. But be warned: as I mentioned above, the code is not super usable right now and it's been so long since I opened the PR that the code might be buggy/stale already.

yuvaltassa, 2024-08-09:
@rottenapple777 if you would clean that up and make it usable, that would be a much-appreciated contribution 👍

rottenapple777, 2024-08-09:
@hartikainen Okay, I'll check it out. Thanks for taking the time to reply.

rohanpsingh, 2024-11-08:
Is there an update on this?