google-deepmind/mujoco_mpc #340: Modifications on Dynamics, Optimiser of Quadrotor

lakshmikanth-nageswar, 2024-08-15:
Hi, 

Could you please tell where the dynamics of the different models [needed for quadrotor specifically] is defined. I am intending to modify the dynamics and looking to implement different controllers in the environment. Could someone guide me on the same?

Thanks!

yuvaltassa, 2024-08-15:
MuJoCo **is** the dynamics. The physics code lives here https://github.com/google-deepmind/mujoco/tree/main/src/engine

closed 2024-08-15 by yuvaltassa

lakshmikanth-nageswar, 2024-08-15:
Thanks for directly to the code. I am looking to modify the constraints, dynamics and implement a different controller in Python for the quadrotor. If I am intending to implement my own controller on the linear dynamics scripted externally, then how to read state data and input control commands from the explicit controller to the environment? Could you please guide a little on that? 

Thanks!

lakshmikanth-nageswar, 2024-08-15:
> MuJoCo **is** the dynamics. The physics code lives here https://github.com/google-deepmind/mujoco/tree/main/src/engine

Thanks for directly to the code. I am looking to modify the constraints, dynamics and implement a different controller in Python for the quadrotor. If I am intending to implement my own controller on the linear dynamics scripted externally, then how to read state data and input control commands from the explicit controller to the environment? Could you please guide a little on that?

Thanks!

yuvaltassa, 2024-08-15:
MuJoCo is open-source and well-documented https://mujoco.readthedocs.io/en/latest/overview.html

Read the docs, change the code however you like 🙂