google-deepmind/mujoco_mpc #343: Support for Hardware-in-the-loop simulation

oscarkfpang, 2024-09-11:
This mjpc is great and I managed to create my own custom robot task and try different cost terms and weights on various planners for simulation. My next step is to include my own real robot and test if the planner's policy can act on the robot under actual environment. 

I know some other people have asked about [collecting trajectories from planners in headless mode](https://github.com/google-deepmind/mujoco_mpc/issues/52) or [updating mjdata with real robot states](https://github.com/google-deepmind/mujoco_mpc/issues/318). I tried but mjpc seems cannot perform accurately when the mjdata is updated by the real robot states data. I think because the mjpc is also running its own physics engine to generate the robot states. And I can't close this loop by exporting the best trajectory back to the real robot. Is there an roadmap for supporting hardware in the loop simulation? And also plans for support in ROS?

Another approach to my problem above is to extract the planner part to run and feed it with the state info from the robot and its sensors to generate trajectories for controlling the robot. I'm experimenting this and if you have any insights please let me know. I hope I can update my progress here.

thowell, 2024-09-25:
Utilizing MJPC for hardware in the loop is possible if you can estimate the MuJoCo state using measurements from your hardware. Probably the easiest way to interface MJPC with the hardware is using the [Python API](https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#python-api).

No plans to support ROS.

yuvaltassa, 2024-09-25:
See also https://arxiv.org/abs/2409.10469