google-deepmind/mujoco_mpc #345: error running ./mjpc

lrchit, 2024-09-25:
Hi,

Thanks for your incredible work of mujoco_mpc! I'm trying to use it on my Ubuntu 20.04, but after every thing went well like this:
[image]
I ran cd /bin and ./mjpc and went across this:
[image]
with mujoco screen:
[image]

Could anyone tell me what happened there? Thanks in advance!

Best, 
Ruochen

yuvaltassa, 2024-09-26:
Unless you are saying that you ran MJPC at HEAD, in which case we might have a model which does not comply with this requirement, in which case we have a bug 🙂 

cc @thowell

thowell, 2024-09-26:
@lrchit which task produces this error?

yuvaltassa, 2024-09-26:
@thowell I don't think OP reached the point where they even selected a task. 
IIUC this just happened to them at the launch of the app?

lrchit, 2024-09-26:
> @thowell I don't think OP reached the point where they even selected a task. IIUC this just happened to them at the launch of the app?

Yes, I just ran ./mjpc and the error message appeared. Should the tasks be specified after ./mjpc?

husnoo, 2024-12-25:
> > @thowell I don't think OP reached the point where they even selected a task. IIUC this just happened to them at the launch of the app?
> 
> Yes, I just ran ./mjpc and the error message appeared. Should the tasks be specified after ./mjpc?

Did you figure this out?

asd417, 2025-05-20:
Running into same issue as well. Compiled myself on Linux 24.04 LTS with a separate mujoco 3.2.3 library

asd417, 2025-05-20:
Figured out.

The reason I was getting this issue is because I was compiling ``libmujoco.so.3.2.3`` from the official mujoco 3.2.3 repo and using ``LD_LIBRARY_PATH`` to the compiled library.

Instead, I compiled the ``libmujoco.so.3.2.3`` using the mujoco-src provided in this repo at ``build/_deps/mujoco-src``