google-deepmind/mujoco_mpc #348: How to perform a grasp experiment using robotiq 3f-gripper?

ppap36, 2024-10-28:
It is easy to perform a grasp experiment using 2f_gripper.
[image]

But how to perform the same using 3f-gripper like this?
[image]
How to control the gripper?

thowell, 2024-11-20:
This [manipulation task](https://github.com/google-deepmind/mujoco_mpc/tree/main/mjpc/tasks/manipulation) uses a [Robotiq gripper](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotiq_2f85) attached to a [Franka Emika Panda arm](https://github.com/google-deepmind/mujoco_menagerie/tree/main/franka_emika_panda). This task could be modified to use a different end effector.

ppap36, 2024-11-29:
Thanks a lot