google-deepmind/mujoco_mpc #360: Running MJPC without GUI

kgimbergh, 2024-11-21:
Hi! I am a student at the TU Delft, I use MuJoCo for my research on balanced standing for an exoskeleton.

I use MJPC in C++ with MuJoCo version 3.1.4 on ubuntu 22.04.

To tune the weights in my cost function, I would like to be able to run MJPC without the GUI of MuJoCo. Now I have read that headless rendering is possible, but I have not been able to accomplish it.

The first option I saw was to install OSMesa or EGL and to set MUJOCO_GL to 'osmesa' or 'egl', respectively. Unfortunately nothing changed in doing this and the MuJoCo GUI launched as it would normally.

When trying to disable the launch of the GUI manually in say simulate.cc, I run into segmentation faults.

I have also seen that it might be possible using python, but I would prefer to keep to C++.

Is there anything that I have looked over? Is there a simple way to get headless rendering? Thanks!

yuvaltassa, 2024-11-21:
Hi @kgimbergh,

What you are asking might be possible, but could you please explain your reasoning? **Why** do you want this? Generally speaking, MJPC should be used either online-interactive with a GUI or offline via gRPC (no rendering). Are you thinking of the online or offline use case? If the latter, why do you need rendering at all?

Note that the offline functionality is implemented via gRPC. It is possible to talk gRPC over Python but of course it is also possible via C++.

boo555, 2025-08-21:
Hi @kgimbergh, did you manage to run mjpc without the GUI after all?

My goal is to run a predefined amount of episodes for a specific task using MJPC. Currently, I have implemented this feature inside the transition function of my task and used a GUI button to start the ``experiment". However, I would like to move to cloud computing for running more experiments and ideally I would like to:
- Configure the task, planner and physics parameters via corresponding config files instead of using the GUI.
- Run MJPC with the provided config files without rendering the environment. I expect that what I have implemented in the transition function will log the episode's outcome together with some other metrics and take care of resetting the environment after each episode. One thing I am missing is how to terminate the mjpc app inside the transition function once the desired number of episodes has been completed.

@yuvaltassa do you maybe have any pointers on how to achieve this with MJPC?

yuvaltassa, 2025-08-26:
Why are you not using the gRPC API again?