google-deepmind/mujoco_mpc #363: Question: Why spline_mapping is uesd to update paramerter?

ppap36, 2024-11-29:
Hello! I am a master student studying mjpc.
When I was studying [https://github.com/google-deepmind/mujoco_mpc/tree/main/mjpc/planners/gradient](url) algorithnm, 
I was looking at these codes:
https://github.com/google-deepmind/mujoco_mpc/blob/9bd2037498056a0ab9cf9463ab841e40fef54142/mjpc/planners/gradient/planner.cc
```
    // compute spline mapping linear operator
    mappings[policy.representation]->Compute(
        candidate_policy[0].times, candidate_policy[0].num_spline_points,
        trajectory[0].times.data(), trajectory[0].horizon - 1);

    // compute total derivatives
    mju_mulMatTVec(candidate_policy[0].parameter_update.data(),
                   mappings[policy.representation]->Get(),
                   candidate_policy[0].k.data(),
                   model->nu * (trajectory[0].horizon - 1),
                   model->nu * candidate_policy[0].num_spline_points);

```

From my point of view, I assume `k` and `parameter_update` are the same thing, and i don't know why `mappings` are used.

Can someone have an idea? Thanks a lot!

thowell, 2024-12-13:
A mapping is used to transform spline points (m x N_s) optimized by the gradient planner to a trajectory of actions (m x N_a). Please see the comment for [spline mapping](https://github.com/google-deepmind/mujoco_mpc/blob/9bd2037498056a0ab9cf9463ab841e40fef54142/mjpc/planners/gradient/spline_mapping.h#L29).

ppap36, 2024-12-14:
I get it, thanks a lot!

closed 2024-12-14 by ppap36