google-deepmind/mujoco_mpc #373: Build Failing

Mehul0x, 2025-02-25:
After Configuring the build, I run the command 

```
cmake --build . --config Release
``` 
There are 3092 steps I think, it fails on around 3086 every time 

I tried reconfiguring and building again, deleting build and starting over, yet I get the same error

I am on ubuntu 20.04.6

> (mjpc) aero@AerosheeeeshUbuntu:~/mujoco_mpc/build$ cmake --build . --config Release 
[0/2] Re-checking globbed directories...
[2/8] Copying Menagerie and dm_control assets into binary directory
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/acrobot/acrobot_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/acrobot/acrobot.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/cartpole/cartpole_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/cartpole/cartpole.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/humanoid/humanoid_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/humanoid/humanoid.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/particle/particle_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/particle/particle.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/swimmer/swimmer_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/swimmer/swimmer.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/walker/walker_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/walker/walker.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/allegro/right_hand_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/allegro/right_hand.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/panda/panda_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/panda/panda.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/quadruped/a1_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/quadruped/a1.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/quadrotor/quadrotor_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/quadrotor/quadrotor.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/common_assets/cube_modified_shadow_reorient.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/common_assets/reorientation_cube.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/rubik/cube_3x3x3_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/rubik/cube_3x3x3.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/rubik/transition_model.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/rubik/cube_3x3x3.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/aloha_cartesian.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/aloha.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/mjx_aloha.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/aloha.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/mjx_scene.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/scene.xml)
Hunk #1 succeeded at 1 with fuzz 1.
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/mjx_filtered_cartesian_actuators.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/bimanual/filtered_cartesian_actuators.xml)
patching file /home/aero/mujoco_mpc/build/mjpc/tasks/op3/op3_modified.xml (read from /home/aero/mujoco_mpc/build/mjpc/tasks/op3/op3.xml)
[3/8] Linking CXX executable bin/filter_server
FAILED: bin/filter_server 
: && /usr/bin/clang++-12  -O3 -DNDEBUG -flto=thin   mjpc/grpc/CMakeFiles/filter_server.dir/filter_server.cc.o mjpc/grpc/CMakeFiles/filter_server.dir/filter.grpc.pb.cc.o mjpc/grpc/CMakeFiles/filter_server.dir/filter.pb.cc.o mjpc/grpc/CMakeFiles/filter_server.dir/filter_service.cc.o  -o bin/filter_server  -Wl,-rpath,/home/aero/mujoco_mpc/build/lib  lib/libabsl_flags.a  lib/libabsl_flags_parse.a  lib/libabsl_status.a  lib/libabsl_strings.a  lib/libgrpc++_reflection.a  lib/libgrpc++.a  lib/libprotobuf.a  lib/libmjpc.a  lib/libabsl_flags_usage.a  lib/libabsl_flags_usage_internal.a  lib/libabsl_leak_check.a  lib/libabsl_die_if_null.a  lib/libabsl_log_initialize.a  lib/libutf8_validity.a  lib/libgrpc.a  lib/libabsl_statusor.a  lib/libupb_collections_lib.a  lib/libupb_json_lib.a  lib/libupb_textformat_lib.a  lib/libutf8_range_lib.a  lib/libupb.a  lib/libre2.a  /usr/lib/x86_64-linux-gnu/libz.so  _deps/grpc-build/third_party/cares/cares/lib/libcares.a  lib/libgpr.a  lib/libabsl_flags.a  lib/libabsl_flags_internal.a  lib/libabsl_flags_reflection.a  lib/libabsl_flags_config.a  lib/libabsl_flags_program_name.a  lib/libabsl_flags_private_handle_accessor.a  lib/libabsl_flags_commandlineflag.a  lib/libabsl_flags_commandlineflag_internal.a  lib/libabsl_status.a  lib/libabsl_flags_marshalling.a  lib/libssl.a  lib/libcrypto.a  lib/libaddress_sorting.a  -lm  -lrt  lib/libabsl_log_internal_check_op.a  lib/libabsl_log_internal_conditions.a  lib/libabsl_log_internal_message.a  lib/libabsl_strerror.a  lib/libabsl_log_internal_nullguard.a  lib/libabsl_examine_stack.a  lib/libabsl_log_internal_format.a  lib/libabsl_log_internal_proto.a  lib/libabsl_log_internal_log_sink_set.a  lib/libabsl_log_internal_globals.a  lib/libabsl_log_globals.a  lib/libabsl_log_sink.a  lib/libabsl_log_entry.a  lib/libabsl_random_distributions.a  lib/libabsl_random_seed_sequences.a  lib/libabsl_random_internal_pool_urbg.a  lib/libabsl_random_internal_randen.a  lib/libabsl_random_internal_randen_hwaes.a  lib/libabsl_random_internal_randen_hwaes_impl.a  lib/libabsl_random_internal_randen_slow.a  lib/libabsl_random_internal_platform.a  lib/libabsl_random_internal_seed_material.a  lib/libabsl_random_seed_gen_exception.a  lib/libabsl_cord.a  lib/libabsl_cordz_info.a  lib/libabsl_cord_internal.a  lib/libabsl_cordz_functions.a  lib/libabsl_cordz_handle.a  lib/libabsl_crc_cord_state.a  lib/libabsl_crc32c.a  lib/libabsl_str_format_internal.a  lib/libabsl_crc_internal.a  lib/libabsl_crc_cpu_detect.a  lib/libabsl_raw_hash_set.a  lib/libabsl_hash.a  lib/libabsl_bad_optional_access.a  lib/libabsl_bad_variant_access.a  lib/libabsl_city.a  lib/libabsl_low_level_hash.a  lib/libabsl_hashtablez_sampler.a  lib/libabsl_exponential_biased.a  lib/libabsl_synchronization.a  lib/libabsl_stacktrace.a  lib/libabsl_symbolize.a  lib/libabsl_debugging_internal.a  lib/libabsl_demangle_internal.a  lib/libabsl_graphcycles_internal.a  lib/libabsl_kernel_timeout_internal.a  lib/libabsl_time.a  lib/libabsl_strings.a  lib/libabsl_string_view.a  lib/libabsl_strings_internal.a  lib/libabsl_int128.a  lib/libabsl_civil_time.a  lib/libabsl_time_zone.a  lib/libabsl_malloc_internal.a  lib/libabsl_throw_delegate.a  lib/libglfw3.a  /usr/lib/x86_64-linux-gnu/librt.so  -lm  -ldl  /usr/lib/x86_64-linux-gnu/libX11.so  lib/liblodepng.a  lib/libmujoco.so.3.2.3  lib/libthreadpool.a  lib/libabsl_base.a  lib/libabsl_raw_logging_internal.a  -pthread  lib/libabsl_log_severity.a  lib/libabsl_spinlock_wait.a  -lrt && :
/usr/bin/ld: lib/libcrypto.a: error adding symbols: file format not recognized
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[4/8] Linking CXX executable bin/direct_server
FAILED: bin/direct_server 
: && /usr/bin/clang++-12  -O3 -DNDEBUG -flto=thin   mjpc/grpc/CMakeFiles/direct_server.dir/direct_server.cc.o mjpc/grpc/CMakeFiles/direct_server.dir/direct.grpc.pb.cc.o mjpc/grpc/CMakeFiles/direct_server.dir/direct.pb.cc.o mjpc/grpc/CMakeFiles/direct_server.dir/direct_service.cc.o  -o bin/direct_server  -Wl,-rpath,/home/aero/mujoco_mpc/build/lib  lib/libabsl_flags.a  lib/libabsl_flags_parse.a  lib/libabsl_status.a  lib/libabsl_strings.a  lib/libgrpc++_reflection.a  lib/libgrpc++.a  lib/libprotobuf.a  lib/libmjpc.a  lib/libabsl_flags_usage.a  lib/libabsl_flags_usage_internal.a  lib/libabsl_leak_check.a  lib/libabsl_die_if_null.a  lib/libabsl_log_initialize.a  lib/libutf8_validity.a  lib/libgrpc.a  lib/libabsl_statusor.a  lib/libupb_collections_lib.a  lib/libupb_json_lib.a  lib/libupb_textformat_lib.a  lib/libutf8_range_lib.a  lib/libupb.a  lib/libre2.a  /usr/lib/x86_64-linux-gnu/libz.so  _deps/grpc-build/third_party/cares/cares/lib/libcares.a  lib/libgpr.a  lib/libabsl_flags.a  lib/libabsl_flags_internal.a  lib/libabsl_flags_reflection.a  lib/libabsl_flags_config.a  lib/libabsl_flags_program_name.a  lib/libabsl_flags_private_handle_accessor.a  lib/libabsl_flags_commandlineflag.a  lib/libabsl_flags_commandlineflag_internal.a  lib/libabsl_status.a  lib/libabsl_flags_marshalling.a  lib/libssl.a  lib/libcrypto.a  lib/libaddress_sorting.a  -lm  -lrt  lib/libabsl_log_internal_ch [... cut ...] 4-linux-gnu/libX11.so  lib/liblodepng.a  lib/libmujoco.so.3.2.3  lib/libthreadpool.a  lib/libabsl_base.a  lib/libabsl_raw_logging_internal.a  -pthread  lib/libabsl_log_severity.a  lib/libabsl_spinlock_wait.a  -lrt && :
/usr/bin/ld: lib/libcrypto.a: error adding symbols: file format not recognized
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[6/8] Building CXX object mjpc/grpc/CMakeFiles/agent_server.dir/agent_server.cc.o
ninja: build stopped: subcommand failed.

Jdvakil, 2025-02-26:
I am having the same build error on multiple ubuntu 20.04 devices, I have been following the instructions in the readme.

yongpeng2001, 2025-02-26:
I encountered the same issue on Ubuntu 20.04 and found a workaround by running `cmake .. -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_C_COMPILER:STRING=clang-12 -DCMAKE_CXX_COMPILER:STRING=clang++-12`. Note that `-G Ninja` and `-DMJPC_BUILD_GRPC_SERVICE:BOOL=ON` are removed. Just a guess, but this issue might be related to Ninja or gRPC.

Mehul0x, 2025-02-26:
I thought gRPC was important?

Jdvakil, 2025-02-26:
The hack worked for building mjpc but crashes with the same error when installing the python packages

Mehul0x, 2025-02-26:
Ninja isn't the issue and yes, can't install the python API

thowell, 2025-03-12:
is there an issue with the build for macos too or only for ubuntu 20.04?

is there an issue with the build using the vscode instructions: https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode?

shaashou, 2025-03-12:
> is there an issue with the build for macos too or only for ubuntu 20.04?
> 
> is there an issue with the build using the vscode instructions: https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode?

My system is Ubuntu 20.04

shaashou, 2025-03-12:
> is there an issue with the build for macos too or only for ubuntu 20.04?macOS 的构建是否存在问题，还是仅针对 Ubuntu 20.04 存在问题？
> 
> is there an issue with the build using the vscode instructions: https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode?使用 VSCode 指令的构建是否存在问题：https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode？

Still can't compile

shaashou, 2025-03-21:
> > > is there an issue with the build for macos too or only for ubuntu 20.04?macOS 的构建是否存在问题，还是仅针对 Ubuntu 20.04 存在问题？
> > > is there an issue with the build using the vscode instructions: https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode?使用 VSCode 指令的构建是否存在问题：https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#build-and-run-mjpc-gui-application-using-vscode？
> > 
> > 
> > My system is Ubuntu 20.04我的系统是 Ubuntu 20.04
> 
> Try to make sure that your Internet connection is stable, as this process requires file downloads.尝试确保您的 Internet 连接稳定，因为此过程需要下载文件。

I'm sure there's nothing wrong with my network.

ppap36, 2025-03-31:
I solved the problem by following instructions of AI.
①install `openssl`
```
sudo apt-get install libssl-dev  # Ubuntu/Debian
```
②Increase the maximum number of files allowed to be opened by the system
```
ulimit -n 4096
```
③Mandatory use of OpenSSL system
```
cmake .. \
  -DCMAKE_BUILD_TYPE=Release \
  -G Ninja \
  -DCMAKE_C_COMPILER=clang-12 \
  -DCMAKE_CXX_COMPILER=clang++-12 \
  -DMJPC_BUILD_GRPC_SERVICE=ON \
  -DgRPC_SSL_PROVIDER=package  # 强制使用系统 OpenSSL
```